project-chip/connectedhomeip #3386: Figure out how to filter out third_party warnings from code scanning

woody-apple, 2020-10-22:
#### Problem
There are hundreds of code scanning alerts coming from third party dependencies like this:

https://github.com/project-chip/connectedhomeip/security/code-scanning/3123?query=ref%3Arefs%2Fheads%2Fmaster

bluez
cirque
happy
pigweed
mbledtls
nrfxlib
nrfconnect
openthread

 #### Proposed Solution
Need to figure out how to disable them properly so this doesn't continue to pollute the (many) code scanning alerts in CHIP code itself.

woody-apple, 2020-10-22:
@mspang could you take a peek at this? 

I took a stab after reading here:
https://docs.github.com/en/enterprise-server@2.22/github/finding-security-vulnerabilities-and-errors-in-your-code/configuring-the-codeql-workflow-for-compiled-languages

But clearly missing something! TLDR, need to clean up the binary files before we run our analysis... (I think?)

woody-apple, 2021-01-27:
ping @mspang, any chance of taking a peek? Our pack of security bugs from Github now is near impossible to parse from all the third party stuff.

mspang, 2021-02-05:
The way this integrates is pretty sketchy. GitHub really should give us a way to drive the scanning ourselves. I understand why they're doing things the way they are - it wants to work out of the box with a lot of build systems. But if we don't drive it ourselves then we don't have fine grained control.

We could try building all 3p code first, then activate CodeQL, then build 1st party code. This will probably remove at least some first party code from CodeQL, though - any case where third party code depends on 1st party code will also be silenced. These cases aren't that common, but can happen - think configuration headers, or other cases of build time dependency-injection. Alternatively, we could try to undo their environment hackery when compiling 3p code.

I think part of the problem is just that a lot of CodeQL is false positives, or at least low signal, and thus not worth fixing. Third party code can have real problems, too, and diagnostics with high signal and severity in third party code would absolutely be worth fixing.

mspang, 2021-02-05:
I'm seeing many many uninteresting diagnostics that are actually from the standard library. Those aren't going to be fixed by splitting the build up into phase, since they are merely #includes from first party code.

What we need is a way to filter by source path, and I don't see anything in the docs that would accomplish that for compiled code.

closed 2022-12-20 by stale[bot]