project-chip/connectedhomeip #3398: Figure out how QR code parsing can be used safely with the MemoryInit requirement

bzbarsky-apple, 2020-10-22:
#### Problem
QR code parsing was changes to use `Platform::Memory` bits, but that requires a `MemoryInit` call.  Unfortunately, scanning and parsing of a QR code can happen independently of overall CHIP stack bringup.  This was mentioned as a possible issue in https://github.com/project-chip/connectedhomeip/pull/3397#issue-508433136 and we ran into this exact issue in the iOS CHIPTool as well.

 #### Proposed Solution
Allow multiple calls to `MemoryInit`, with the ones after the first call no-ops except for incrementing a counter.  Have `MemoryShutdown` decrement the counter; in case it ever becomes nontrivial, the nontrivial actions will be taken only when the counter goes to 0.  Once that's done, the QR code parsing bits can init and shutdown around whatever work they need to do.

Alternately, we could go back to not using `Platform::Memory` for QR code parsing, since that's not going to be running on constrained devices....

@andy31415 thoughts?
@sagar-apple

bzbarsky-apple, 2020-10-22:
@kpschoedel also.  https://github.com/project-chip/connectedhomeip/pull/3260 talks about "embedded device-side code", which `src/setup_payload/QRCodeSetupPayloadParser.cpp` doesn't seem to be.

kpschoedel, 2020-10-22:
This is one reason I don't really like CHIPMem.h — it makes it harder to combine CHIP-library code with non-CHIP code.

That being said, my current intent for #3178 (_Use mbedtls_platform_set_calloc_free() for non-malloc() configurations_), after sleeping on it, is
- rename the allocator-implementation MemoryInit() and MemoryShutdown() in CHIPMem-Malloc.cpp and CHIPMem-SimpleAlloc.cpp
- wrap these in an allocator-implementation–independent MemoryInit() and MemoryShutdown() that also performs configuration-specific operators like calling mbedtls_platform_set_calloc_free().
In that case, the generic MemoryInit() and MemoryShutdown() can easily ensure that they call the implementation functions only once.

bzbarsky-apple, 2020-10-22:
Another possible API is to allow arbitrarily many calls to `MemoryInit` but have `MemoryShutdown` shut things down immediately.  This is what `CHIPMem-SimpleAlloc` seems to implement in practice....

bzbarsky-apple, 2020-10-22:
@kpschoedel That would work for the use cases here, I think.  @sagar-apple pointed out that at least for the "show QR code on screen" use case the device does need to use `src/setup_payload/QRCodeSetupPayloadGenerator.cpp`, which presumably ends up poking at memory bits via TLV.

kpschoedel, 2020-10-22:
> Another possible API is to allow arbitrarily many calls to `MemoryInit` but have `MemoryShutdown` shut things down immediately. This is what `CHIPMem-SimpleAlloc` seems to implement in practice....

Assuming it gets wrapped anyway, the allocator implementations can assume no nested calls, and the specification of MemoryShutdown can be independent of the allocator implementation.

PR #3421 (2020-10-26): Allow nesting Platform::MemoryInit and MemoryShutdown
#### Problem

Some code wants to use `CHIPMem.h` allocation in contexts where
higher-level code might or might not already have initialized it.
Allowing multiple calls (proposed in issue #3398) solves this.

#### Summary of Changes

- Rename the allocator-specific `MemoryInit()` and `MemoryShutdown()`.
- Call those renamed functions from an allocator-independent
  `MemoryInit()` and `MemoryShutdown()` that allow mulitple calls
  and invoke the underlying allocator only once.

Fixes #3398 — Figure out how QR code parsing can be used safely w [... cut]
commit ac8a2e7a (2020-10-26): Allow nesting Platform::MemoryInit and MemoryShutdown (#3421)

#### Problem

Some code wants to use `CHIPMem.h` allocation in contexts where
higher-level code might or might not already have initialized it.
Allowing multiple calls (proposed in issue #3398) solves this.

#### Summary of Changes

- Rename the allocator-specific `MemoryInit()` and `MemoryShutdown()`.
- Call those renamed functions from an allocator-independent
  `MemoryInit()` and `MemoryShutdown()` that allow mulitple calls
  and invoke the underlying allocator only once.

Fixes #3398 — Figure out how QR code parsing can be used [... cut]
closed 2020-10-26 by BroderickCarlin