project-chip/connectedhomeip #3401: [ES32] stops advertising on unsubscribe when built with NimBLE

mspang, 2020-10-22:
repro: `chip-tool echo ble 3840 12345678`

Then terminate chip-tool to unsubscribe.

```
I (42415) chip[DL]: CHIPoBLE unsubscribe received
I (42425) chip[DL]: BLE GAP connection terminated (con 32387
I (42965) chip[BLE]: Releasing end point's BLE connection back to application.
E (42965) chip[DL]: Long dispatch time: 543 ms
I (42965) chip[DL]: Setup PIN discriminator not found; using default: f00
I (42975) chip[DL]: Configuring CHIPoBLE advertising (interval 200 ms, connectable, device name CHIP-3840)
E (42985) chip[DL]: ble_gap_adv_start() failed: Error 2 (0x00000002)
E (42985) chip[DL]: Configure Adv Data failed: Error 2 (0x00000002)
E (42995) chip[DL]: Disabling CHIPoBLE service due to error: Error 2 (0x00000002)
```

Device stops advertising here, instead of restarting advertising with 200ms interval.

@prasad-alatkar

mspang, 2020-10-22:
This is a regression in the NimBLE integration that's not present in the Bluedroid integration.

mspang, 2020-10-23:
@kedars

prasad-alatkar, 2020-10-23:
Hi @mspang I will take a look at this and update.

closed 2020-11-10 by mspang