project-chip/connectedhomeip #3407: echo-ble triggers network provisioning with invalid ssid/password

mspang, 2020-10-23:
Repro: on a mac, `chip-tool echo ble 3840 12345678`
```
I (892834) rendezvous-devicedelegate: Device completed SPAKE2+ handshake

I (892834) wifi-echo-demo: Display awake but will switch off automatically in 30 seconds
I (893874) chip[NP]: Received kWiFiAssociationRequest. DeviceDelegate 0x3ffe5fa0

I (893874) chip[NP]: ESP32NetworkProvisioningDelegate: SSID: , key: 
```

@pan-apple
@andy31415

pan-apple, 2020-10-23:
Tag: @vivien-apple

vivien-apple, 2020-10-27:
@pan-apple `echo` capabilities over BLE has been removed in your cleanup right ? 

I was planning to convert `echo ble` to 'pair ble' at some point so that's fine.
I haven't looked in details, is there something like `PersistentStorageDelegate` for the device side ? Or is it up to the demo app at the moment ?

woody-apple, 2021-01-27:
@vivien-apple is this still relevant?

vivien-apple, 2021-02-05:
Nope. The `echo` command was handcrafted and it does not exists anymore.

closed 2021-02-05 by vivien-apple