project-chip/connectedhomeip #3411: Setup automated chip-build docker image builds

gjc13, 2020-10-23:
#### Problem
Currently we need to ask authorized maintainers personally to upload the`chip-build-xxx` docker images.

 #### Proposed Solution
Refer to https://docs.docker.com/docker-hub/builds/ to setup automated builds and tag latest.

mspang, 2020-10-25:
We're running the build & tests on GitHub actions, not docker cloud. So building the images there is what I was thinking as it seems like a more natural fit. Maybe we could use

https://github.com/marketplace/actions/build-and-push-docker-images

I haven't looked at it in depth but I agree improving this process would help PRs that change docker images move along much more quickly.

@rwalker-apple @woody-apple

andy31415, 2022-01-31:
Existing setup works for now - not ideal, however due to support of versions, we are likely to sabilize build images. Will not build this as a bugfix for v1.

closed 2023-01-14 by stale[bot]