project-chip/connectedhomeip #3416: Come up with proposal to flag PRs that increase Flash/RAM

shana-apple, 2020-10-23:
We need to ensure that Flash + RAM usage does not go up when it shouldn't. We need to come up with a process in CI that flags PRs that  increase Flash/RAM so that the originator of the PR can explain why that should be allowed or if he should fix it.

Come up with proposal.

kpschoedel, 2020-10-23:
Sub-issue #3415 (Try to detect new dynamic heap usage at build time) is probably the hardest, if we want to do that. Run-time use is easy to measure but the problem becomes having a run actually exercise a new PR.

For stack, maybe we want `-fstack-usage` analysis and/or `-Wstack-usage`.

For flash and statically-allocated RAM we can get numbers from tools like bloaty (or nm, size, etc.). Perhaps we should have an explicit indication if an increase is disproportionate to the amount of code in a PR.

andy31415, 2020-10-23:
Flash usage may be covered by our bloat report (which may need some fixing ... maybe we should only run it for a subset of minimal apps so that we can have more history and it works on more PRs) and so is BSS.

Heap on the other hand is hard. Generally I would expect minimal app to use no heap at all. Maybe we could build a demo app as such and then CI validate that 'no heap is used in a minimal lock/sensor app'. Building a end-to-end test that does everything (rendezvous, SC, communication, eventing) does seem a bit involved.

kpschoedel, 2020-10-23:
If we get to a minimal app using no heap, then it shouldn't be too difficult to use linker or post-linker games on at least one platform to make sure it stays that way, e.g. make sure debug symbols don't contain `malloc`.

closed 2022-10-14 by stale[bot]