project-chip/connectedhomeip #3417: Investigate .bss usage from ot::Instance

kpschoedel, 2020-10-23:
#### Problem

ot::Instance in third_party/openthread/repo/src/core/common/instance.cpp has a .bss segment around 48K.

 #### Proposed Solution

I suspect we may be building with unnecessary openthread options — possibly the 48K is a redundant internal heap.

bukepo, 2020-10-26:
By default, OpenThread has two big memory pools.

1. The message pool, this allocates about 20K(`160 * 32 * 4=20480`) on nRF52840 platform. This is only used by OpenThread itself.
2. The mbedtls heap, this allocates about 16K on nRF52840 platform. This is shared by all components uses mbedTLS.

We'll investigate whether we can enhance the first memory pool.

erjiaqing, 2020-10-30:
The nRF5 example app should use less bss memory after #3454 is merged, as for the nrfconnect, there is a pending upstream change to be merged to nrfconnect sdk.

https://github.com/zephyrproject-rtos/zephyr/commit/5b7efd32777b8da15ab18dc6b9f711635af20baa and https://github.com/zephyrproject-rtos/zephyr/commit/102084f873a5b8af1b88385df2ce6ed6f9390833 

@LuDuda will work on cherry-picking these commits to nrfconnect, and will proceed after that.

erjiaqing, 2021-01-28:
I think it still needs more attempts before this issue is fully resolved, maybe we can mark this issue as low priority for now?

woody-apple, 2021-01-29:
@erjiaqing Can we update where we are memory wise, given it sounds like we've made progress? Makes sense to lower priority if the remaining size is small!

mrjerryjohns, 2022-01-31:
Closing issue for now, and opening if OT size is still a concern.

closed 2022-01-31 by mrjerryjohns