project-chip/connectedhomeip #3418: src/app/encoder.cpp is a monolith using excessive flash

kpschoedel, 2020-10-23:
#### Problem

src/app/encoder.cpp seems to contain all possible clusters, and is the single largest use of flash in example apps.

 #### Proposed Solution

Split off individual parts. The file also has some ‘forced inlining’ using C macros that increase code size.

kpschoedel, 2020-10-26:
This is not a practical problem for common toolchains (gcc, clang) that support `-ffunction-sections` and `-Wl,--gc-sections`.

bzbarsky-apple, 2020-10-27:
Yes, we should be building with `-ffunction-sections` everywhere.  If we're not, for some reason, a lot of things will be bad, starting with the various unused functions in the Silicon Labs code that we imported suddenly being compiled in...

kpschoedel, 2020-10-28:
I've definitely seen this in the past, but I don't see it anywhere now.

closed 2020-10-28 by kpschoedel