project-chip/connectedhomeip #3423: BDX State Machine and Message Definitions

holbrookt, 2020-10-23:
#### Problem
Need to implement main business logic and state machine for the BDX spec.

 #### Proposed Solution
Implement a state machine for BDX message exchanges according to spec. This includes logic on what messages can be received at a specific point in a transfer, depending on the parameters of the transfer (sender vs. receiver drive, async vs sync, max length, etc). At minimum, this will also require declaring message types in the BDX "profile".

This should include unit tests to ensure the state machine behaves according to spec.

holbrookt, 2020-11-18:
The merged PR was just for message definitions. PR for state machine logic is coming later

BroderickCarlin, 2020-11-20:
Looks like we got bit by some automations. Sorry about that!

PR #4473 (2021-02-05): BDX TransferSession spec implementation
 #### Problem
The Bulk Data Transfer spec needs to be implemented. There should be an API for beginning and managing a single transfer session, which enforces the rules of the spec.

 #### Summary of Changes
* Add a **TransferSession** class which provides methods for beginning and managing a transfer session. 
* Add test file with at least one case that tests a full session from beginning to end.
* some small changes to enum naming in BdxMessages.h

 Fixes #3423
closed 2021-02-05 by andy31415