project-chip/connectedhomeip #3441: Try building with link-time optimization

kpschoedel, 2020-10-26:
#### Problem

Unused code should not take up flash.

There is at least one large case (#3418) and likely a number of smaller cases where a single source file contains both function(s) that are required and function(s) that are not required but included due to linking granularity being per-file. Link-time optimization (in at least gcc- and llvm-based tool chains) might be a workable alternative to source-level refactoring.

 #### Proposed Solution

Build the minimal app (#2762) with `-flto`. If the results are good, consider applying LTO wherever possible.

kpschoedel, 2020-10-27:
Building with `-flto -fwhole_program` has almost zero effect (0.1%) on nrf5_lighting_app. This seems odd; perhaps something else is inhibiting useful LTO.

Damian-Nordic, 2020-10-28:
> Building with `-flto -fwhole_program` has almost zero effect (0.1%) on nrf5_lighting_app. This seems odd; perhaps something else is inhibiting useful LTO.

Isn't it because we already build with `-ffunction-sections` and `--gc-sections`? `-flto` might still be useful to optimize out parts of a function which are never executed by a given app (e.g. certain function arguments are never used), but maybe there are no many such places (?).

kpschoedel, 2020-10-28:
`--gc-sections` certainly overlaps some of what LTO could do, but there is at least some clear low-hanging fruit in the binary (e.g. calls to functions where the body is strictly smaller than the call instruction) that makes me think LTO is not operating, even though `-v` on the link shows it's being invoked.

nikita-s-wrk, 2021-05-28:
We've tried using LTO for `qpg6100` and it resulted in pretty noticeable size savings:

Stock:
```
+----------------+-------------------------------+
|                | chip-qpg6100-lighting-example |
+----------------+-------------------------------+
| Flash          | 454268/524288 - 86.64 %       |
| Flash+NVM      | 470652/524288 - 89.77 %       |
| Ram            | 59300/98272 - 60.34 %         |

```

With LTO:
`  cflags = [ "-Os", "-flto", "-ffat-lto-objects" ]`

```
+----------------+-------------------------------+
|                | chip-qpg6100-lighting-example |
+----------------+-------------------------------+
| Flash          | 419672/524288 - 80.05 %       |
| Flash+NVM      | 436056/524288 - 83.17 %       |
| Ram            | 58960/98272 - 60.00 %         |

```

However, build flags controlled by pigweed project, should we deliver LTO support to pigweed first?

closed 2022-12-20 by stale[bot]