project-chip/connectedhomeip #3493: Add a dedicated callback to DevicePairingDelegate for requesting Thread credentials

vidhis88, 2020-10-28:
#### Problem
Currently DevicePairingDelegate has OnNetworkCredentialsRequested callback. This is for requesting Wi-Fi network credentials.
However, another one is required to request Thread credentials.

(From chat with @pan-apple: The RendezvousSession/NetworkProvisioning would know which one to call - OnNetworkCredentialsRequested vs OnThreadCredentialsRequested, based on the message type.)

 #### Proposed Solution
Add a distinct callback - OnThreadCredentialsRequested, to DevicePairingDelegate.

woody-apple, 2021-03-09:
@sagar-apple Any thoughts here?

sagar-apple, 2021-03-09:
> Currently DevicePairingDelegate has OnNetworkCredentialsRequested callback. This is for requesting Wi-Fi network credentials.
However, another one is required to request Thread credentials.

I think rather than creating two types of callbacks we could just use our existing API as an indication that the accessory is read to be commissioned with network credentials.  
We could do one of the following things once we have the network provisioning cluster enabled.

1. The SDK layer can implicitly check which type of credentials are supported/needed and call into `onNetworkCredentialsRequested` with a flag/bitmask indicating the type of credentials needed. The Commissioner can then call into the network provisioning cluster and supply the required credentials.

2. The commissioner side can check which credentials the accessory supports/needs and pass those in when the SDK callback `onNetworkCredentialsRequested` is triggered. That would mean our callback API remains Credential agnostic. Just an indication that the accessory is ready to receive credentials.