project-chip/connectedhomeip #3517: Investigate .bss usage of gRendezvousServer

shana-apple, 2020-10-29:
1.1%  5.95Ki    ../../third_party/connectedhomeip/examples/common/chip-app-server/Server.cpp
    79.6%  4.73Ki    .bss
      37.1%  1.76Ki    (anonymous namespace)::gRendezvousServer

shana-apple, 2020-10-29:
(gdb) ptype /o chip::RendezvousServer
/* offset    |  size */  type = class chip::RendezvousServer : public chip::RendezvousSessionDelegate {
                         private:
/*    8      |  1776 */    class chip::RendezvousSession : public chip::SecurePairingSessionDelegate, public chip::RendezvousSessionDelegate, public chip::RendezvousDeviceCredentialsDelegate, public chip::NetworkProvisioningDelegate {
                             private:
/* XXX  8-byte hole */
/*   24      |     4 */        chip::Transport::Base *mTransport;
/*   28      |     4 */        chip::RendezvousSessionDelegate *mDelegate;
/*   32      |    32 */        class chip::RendezvousParameters {
                                 private:
/*   32      |    16 */            class chip::Optional<unsigned long long> [with T = unsigned long long] {
                                     private:
/*   32      |     8 */                T mValue;
/*   40      |     1 */                bool mHasValue;
/* XXX  7-byte padding */
                                       /* total size (bytes):   16 */
                                   } mLocalNodeId;
/*   48      |     4 */            uint32_t mSetupPINCode;
/*   52      |     2 */            uint16_t mDiscriminator;
/* XXX  2-byte hole */
/*   56      |     4 */            chip::Ble::BleLayer *mBleLayer;
/*   60      |     2 */            uint16_t mConnectionObject;
/* XXX  2-byte padding */
                                   /* total size (bytes):   32 */
                               } mParams;
/*   64      |  1672 */        class chip::SecurePairingSession {
                                 private:
                                   static const size_t kSpake2p_WS_Length;
/*   68      |     4 */            chip::SecurePairingSessionDelegate *mDelegate;
/*   72      |     1 */            chip::SecurePairingSession::Spake2pMsgType mNextExpectedMsg;
/* XXX  3-byte hole */
/*   76      |  1432 */            class chip::Crypto::Spake2p_P256_SHA256_HKDF_HMAC : public chip::Crypto::Spake2p {
                                     private:
/* XXX 140-byte hole */
/*  228      |   256 */                class chip::Crypto::Hash_SHA256_stream {
                                         private:
/*  228      |   256 */                    struct chip::Crypto::HashSHA256OpaqueContext {
/*  228      |   256 */                        uint8_t mOpaque[256];
                                               /* total size (bytes):  256 */
                                           } mContext;
                                           /* total size (bytes):  256 */
                                       } sha256_hash_ctx;
/*  484      |  1024 */                struct chip::Crypto::Spake2pOpaqueContext {
/*  484      |  1024 */                    uint8_t mOpaque[1024];
                                           /* total size (bytes): 1024 */
                                       } mSpake2pContext;
                                       /* total size (bytes): 1432 */
                                   } mSpake2p;
/* 1508      |    65 */            uint8_t mPoint[65];
/* 1573      |    80 */            uint8_t mWS[2][40];
/* 1653      |    32 */            uint8_t mKe[32];
/* XXX  3-byte hole */
/* 1688      |     4 */            size_t mKeLen;
/* 1692      |     1 */            bool mPairingComplete;
                                 protected:
/* XXX  3-byte hole */
/* 1696      |    16 */            class chip::Optional<unsigned long long> [with T = unsigned long long] {
                                     private:
/* 1696      |     8 */                T mValue;
/* 1704      |     1 */                bool mHasValue;
/* XXX  7-byte padding */
--Type <RET> for more, q to quit, c to continue without paging--
                                       /* total size (bytes):   16 */
                                   } mLocalNodeId;
/* 1712      |    16 */            class chip::Optional<unsigned long long> [with T = unsigned long long] {
                                     private:
/* 1712      |     8 */                T mValue;
/* 1720      |     1 */                bool mHasValue;
/* XXX  7-byte padding */
                                       /* total size (bytes):   16 */
[... 8 lines cut ...]
/* 1736      |     4 */            chip::NetworkProvisioningDelegate *mDelegate;
/* 1740      |    16 */            class chip::Inet::IPAddress {
                                     public:
/* 1740      |    16 */                uint32_t Addr[4];
                                       static chip::Inet::IPAddress Any;
                                       /* total size (bytes):   16 */
                                   } mDeviceAddress;
                                   /* total size (bytes):   20 */
                               } mNetworkProvision;
/* 1756      |    17 */        class chip::SecureSession {
                                 private:
                                   static const size_t kAES_CCM128_Key_Length;
/* 1756      |     1 */            bool mKeyAvailable;
/* 1757      |    16 */            uint8_t mKey[16];
                                   /* total size (bytes):   17 */
                               } mSecureSession;
/* XXX  3-byte hole */
/* 1776      |     4 */        uint32_t mSecureMessageIndex;
/* 1780      |     2 */        uint16_t mNextKeyId;
/* 1782      |     1 */        chip::RendezvousSession::State mCurrentState;
/* XXX  1-byte padding */
                               /* total size (bytes): 1776 */
                           } mRendezvousSession;
                           /* total size (bytes): 1784 */
                         }

shana-apple, 2020-10-29:
`(gdb) ptype /o chip::RendezvousServer
/* offset    |  size */  type = class chip::RendezvousServer : public chip::RendezvousSessionDelegate {
                         private:
/*    8      |  1776 */    class chip::RendezvousSession : public chip::SecurePairingSessionDelegate, public chip::RendezvousSessionDelegate, public chip::RendezvousDeviceCredentialsDelegate, public chip::NetworkProvisioningDelegate {
                             private:
/* XXX  8-byte hole */
/*   24      |     4 */        chip::Transport::Base *mTransport;
/*   28      |     4 */        chip::RendezvousSessionDelegate *mDelegate;
/*   32      |    32 */        class chip::RendezvousParameters {
                                 private:
/*   32      |    16 */            class chip::Optional<unsigned long long> [with T = unsigned long long] {
                                     private:
/*   32      |     8 */                T mValue;
/*   40      |     1 */                bool mHasValue;
/* XXX  7-byte padding */
                                       /* total size (bytes):   16 */
                                   } mLocalNodeId;
/*   48      |     4 */            uint32_t mSetupPINCode;
/*   52      |     2 */            uint16_t mDiscriminator;
/* XXX  2-byte hole */
/*   56      |     4 */            chip::Ble::BleLayer *mBleLayer;
/*   60      |     2 */            uint16_t mConnectionObject;
/* XXX  2-byte padding */
                                   /* total size (bytes):   32 */
                               } mParams;
/*   64      |  1672 */        class chip::SecurePairingSession {
                                 private:
                                   static const size_t kSpake2p_WS_Length;
/*   68      |     4 */            chip::SecurePairingSessionDelegate *mDelegate;
/*   72      |     1 */            chip::SecurePairingSession::Spake2pMsgType mNextExpectedMsg;
/* XXX  3-byte hole */
/*   76      |  1432 */            class chip::Crypto::Spake2p_P256_SHA256_HKDF_HMAC : public chip::Crypto::Spake2p {
                                     private:
/* XXX 140-byte hole */
/*  228      |   256 */                class chip::Crypto::Hash_SHA256_stream {
                                         private:
/*  228      |   256 */                    struct chip::Crypto::HashSHA256OpaqueContext {
/*  228      |   256 */                        uint8_t mOpaque[256];
                                               /* total size (bytes):  256 */
                                           } mContext;
                                           /* total size (bytes):  256 */
                                       } sha256_hash_ctx;
/*  484      |  1024 */                struct chip::Crypto::Spake2pOpaqueContext {
/*  484      |  1024 */                    uint8_t mOpaque[1024];
                                           /* total size (bytes): 1024 */
                                       } mSpake2pContext;
                                       /* total size (bytes): 1432 */
                                   } mSpake2p;
/* 1508      |    65 */            uint8_t mPoint[65];
/* 1573      |    80 */            uint8_t mWS[2][40];
/* 1653      |    32 */            uint8_t mKe[32];
/* XXX  3-byte hole */
/* 1688      |     4 */            size_t mKeLen;
/* 1692      |     1 */            bool mPairingComplete;
                                 protected:
/* XXX  3-byte hole */
/* 1696      |    16 */            class chip::Optional<unsigned long long> [with T = unsigned long long] {
                                     private:
/* 1696      |     8 */                T mValue;
/* 1704      |     1 */                bool mHasValue;
/* XXX  7-byte padding */
--Type <RET> for more, q to quit, c to continue without paging--
                                       /* total size (bytes):   16 */
                                   } mLocalNodeId;
/* 1712      |    16 */            class chip::Optional<unsigned long long> [with T = unsigned long long] {
                                     private:
/* 1712      |     8 */                T mValue;
/* 1720      |     1 */                bool mHasValue;
/* XXX  7-byte padding */
                                       /* total size (bytes):   16 */
[... 8 lines cut ...]
/* 1736      |     4 */            chip::NetworkProvisioningDelegate *mDelegate;
/* 1740      |    16 */            class chip::Inet::IPAddress {
                                     public:
/* 1740      |    16 */                uint32_t Addr[4];
                                       static chip::Inet::IPAddress Any;
                                       /* total size (bytes):   16 */
                                   } mDeviceAddress;
                                   /* total size (bytes):   20 */
                               } mNetworkProvision;
/* 1756      |    17 */        class chip::SecureSession {
                                 private:
                                   static const size_t kAES_CCM128_Key_Length;
/* 1756      |     1 */            bool mKeyAvailable;
/* 1757      |    16 */            uint8_t mKey[16];
                                   /* total size (bytes):   17 */
                               } mSecureSession;
/* XXX  3-byte hole */
/* 1776      |     4 */        uint32_t mSecureMessageIndex;
/* 1780      |     2 */        uint16_t mNextKeyId;
/* 1782      |     1 */        chip::RendezvousSession::State mCurrentState;
/* XXX  1-byte padding */
                               /* total size (bytes): 1776 */
                           } mRendezvousSession;
                           /* total size (bytes): 1784 */
                         }`

shana-apple, 2020-10-29:
Important to look at:

 struct chip::Crypto::Spake2pOpaqueContext {
/*  484      |  1024 */                    uint8_t mOpaque[1024];
                                           /* total size (bytes): 1024 */
                                       } mSpake2pContext;
                                       /* total size (bytes): 1432 */

class chip::Crypto::Hash_SHA256_stream {
                                         private:
/*  228      |   256 */                    struct chip::Crypto::HashSHA256OpaqueContext {
/*  228      |   256 */                        uint8_t mOpaque[256];
                                               /* total size (bytes):  256 */
                                           } mContext;
                                           /* total size (bytes):  256 */
                                       } sha256_hash_ctx;

pan-apple, 2020-10-29:
The current code is using the secret derived by the Pairing Session to construct the secure session. Final solution (for V1) will use operational credentials for secure session construction. The pairing session will not be required at that time. The memory requirement for operational credentials will be significantly lower than the pairing session. So that should resolve this issue.

closed 2022-10-12 by stale[bot]