project-chip/connectedhomeip #3519: Investigate .bss usage of gSessions

shana-apple, 2020-10-29:
1.1%  5.95Ki    ../../third_party/connectedhomeip/examples/common/chip-app-server/Server.cpp
    79.6%  4.73Ki    .bss
      37.1%  1.76Ki    (anonymous namespace)::gRendezvousServer
      34.5%  1.63Ki    (anonymous namespace)::gTestPairing
      28.4%  1.34Ki    (anonymous namespace)::gSessions

pan-apple, 2020-10-29:
This issue is related to memory consumed by `chip::Transport::PeerConnectionState mStates[16];` in `SecureSessionMgrBase` class. The memory consumption is related to number of peer connection states the object is maintaining. The number of peer connections is configurable by `CHIP_CONFIG_PEER_CONNECTION_POOL_SIZE`. So that could be used to reduce the memory consumption. But, if it is not significant amount of memory, maybe we should keep it to the default value.

kpschoedel, 2022-01-31:
Obsolete — `gSessions` doesn't even exist any more.

closed 2022-01-31 by kpschoedel