project-chip/connectedhomeip #3526: [transport] Add unsecure message support to SecureSessionMgr

erjiaqing, 2020-10-30:
#### Problem
Currently, SecureSessionMgr only supports encrypted messages, and assume all packets are encrypted. However, messages like secure pairing, are not encrypted. We need to allow SecureSessionMgr able to process packets that are not encrypted, so we can unify BLE, TCP and UDP packet processing.

pan-apple, 2020-10-30:
CHIP's initial pairing is the only-use case where messages are exchanged without encryption. That use-case is handled separately by using BLE transport directly. I don't think we should add support for unencrypted packets thru `SecureSessionMgr`.

erjiaqing, 2020-11-05:
Since we won't support unsecured messages for secure sessions, close this issue now.

closed 2020-11-05 by erjiaqing