project-chip/connectedhomeip #3539: Figure out what to do with emberSetBinding/emberGetBinding and more generally with ZCL bindings

vivien-apple, 2020-10-30:
#### Problem

The ZCL bindings table is used by some plugins such as the `Reporting` plugin as well as `IAS Zone`.

The binding table contains struct entries that defines a kind of permanent (they are stored as HAL tokens) link between an accessory and a device.

For example one could create a  binding entry such as:
```
EmberBindingTableEntry bindingEntry;
bindingEntry.type      = EMBER_UNICAST_BINDING;
bindingEntry.local     = localEndpoint;
bindingEntry.clusterId = localClusterId;
bindingEntry.remote    = remoteEndpoint;
bindingEntry.nodeId    = ChipNodeId;
```

The reporting plugin use such a binding to know which devices it needs to send a report too by calling `emberAfSendCommandUnicastToBindingsWithCallback` which ends up calling https://github.com/project-chip/connectedhomeip/blob/e7480cde9c6947491da6f82f9527b584a7b24b13/src/app/util/af-main-common.cpp#L506

The ZCL `report` set of commands are used to configure the reporting variables such as intervals, thresholds, etc... And this config will be used for all bindings (AFAIK).

The main issue here is that the current way to set up bindings are using ZDO if I have understood correctly https://docs.silabs.com/zigbee/latest/em35x/group-util-zdo#ga6856235a7e67c51062355eaafd19a65f).

So we need to figure out what to do.

vivien-apple, 2020-10-30:
@rwalker-apple @bzbarsky-apple Do you have any ideas/suggestions ?

bzbarsky-apple, 2020-10-31:
We presumably need to implement https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Descriptor-Cluster.adoc#binding-attribute -- that's the current proposal for how this should work in CHIP.

vivien-apple, 2021-02-16:
This can be closed. I have added preliminary support for that in #4156.

closed 2021-02-16 by vivien-apple