project-chip/connectedhomeip #3548: Chip tool server crash on raspi4 when doing connect using python chip-device-ctrl

cjandhyala, 2020-10-30:
When trying to connect to the chip device on raspi4 

Stack trace from the python device controller
-------------------------------------------------------
python3 chip-device-ctrl
Chip Device Controller Shell

chip-device-ctrl > help

Available commands (type help <name> for more information):
============================================================
ble-adapter-print   ble-disconnect    close                      
ble-adapter-select  ble-scan          connect                    
ble-connect         ble-scan-connect  set-pairing-wifi-credential
ble-debug-log       btp-connect     

chip-device-ctrl > ble-scan-connect CHIP-2976
2020-10-30 11:07:01,030 ChipBLEMgr   INFO     BLE is ready!
2020-10-30 11:07:01,432 ChipBLEMgr   INFO     adding to scan list:
2020-10-30 11:07:01,432 ChipBLEMgr   INFO     
2020-10-30 11:07:01,432 ChipBLEMgr   INFO     Name =    None                                                                            
2020-10-30 11:07:01,432 ChipBLEMgr   INFO     ID =      BF518E80-E89D-47FF-A645-4B2D1D9E27D2                                            
2020-10-30 11:07:01,432 ChipBLEMgr   INFO     RSSI =    -46                                                                             
2020-10-30 11:07:01,433 ChipBLEMgr   INFO     ADV data: {
    kCBAdvDataChannel = 37;
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataServiceData =     {
        FEAF = {length = 3, bytes = 0x000f00};
    };
}
2020-10-30 11:07:01,433 ChipBLEMgr   INFO     
2020-10-30 11:07:05,655 ChipBLEMgr   INFO     adding to scan list:
2020-10-30 11:07:05,655 ChipBLEMgr   INFO     
2020-10-30 11:07:05,655 ChipBLEMgr   INFO     Name =    None                                                                            
2020-10-30 11:07:05,655 ChipBLEMgr   INFO     ID =      C721EBDD-B7EE-4AE0-949B-49481DC4CB2C                                            
2020-10-30 11:07:05,655 ChipBLEMgr   INFO     RSSI =    -56                                                                             
2020-10-30 11:07:05,656 ChipBLEMgr   INFO     ADV data: {
    kCBAdvDataChannel = 37;
    kCBAdvDataIsConnectable = 1;
    kCBAdvDataServiceData =     {
        FEAF = {length = 17, bytes = 0x1001010101000100010000000000000000};
    };
    kCBAdvDataServiceUUIDs =     (
        FEAF
    );
}
2020-10-30 11:07:05,656 ChipBLEMgr   INFO     
2020-10-30 11:07:05,656 ChipBLEMgr   INFO     scanning stopped
2020-10-30 11:07:05,656 ChipBLEMgr   INFO     trying to connect to CHIP-2976
2020-10-30 11:07:08,453 ChipBLEMgr   INFO     Discovering services
2020-10-30 11:07:08,455 ChipBLEMgr   INFO     Discovering services
2020-10-30 11:07:08,615 ChipBLEMgr   INFO     connect success
chip-device-ctrl > set-pairing-wifi-credential testap testtest
WiFi credential set
chip-device-ctrl > connect 12345678
CHIP: [BLE] subscribe complete, ep = 0x114690c88
CHIP: [BLE] connect handshake timed out, closing ep 0x114690c88
CHIP: [BLE] Auto-closing end point's BLE connection.
2020-10-30 11:07:47,426 ChipBLEMgr   INFO     BLE disconnected, error = None

Here is stack trace from the raspi , chip tool server...
------------------------------------------------------
<suggested fix, suggested enhancement>
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x10, 0x01, 0x01, 0x01, 0x01, 0x00, 0x01, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00]>}
CHIP:DL: HandlePlatformSpecificBLEEvent , 16388
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent , 16389
[Thread 0xfffff56cefc0 (LWP 2440211) exited]

[... 57 lines cut ...]
#7  0x0000fffff7c8d058 in g_closure_invoke () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#8  0x0000fffff7ca1c74 in  () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#9  0x0000aaaaaab6cfec in _bluez_gatt_characteristic1_skeleton_handle_method_call
    (connection=0xaaaaaacc3040, sender=0xffffe802b580 ":1.5", object_path=0xffffe802d210 "/chipoble/3c0d/service/c1", interface_name=0xffffe802c090 "org.bluez.GattCharacteristic1", method_name=0xffffe802c070 "AcquireWrite", parameters=0xffffe8004af0, invocation=0xffffe802c480, user_data=0xffffe4003b50) at gen/platform/Linux/dbus/bluez/DbusBluez.c:12258
#10 0x0000fffff7e210e8 in  () at /lib/aarch64-linux-gnu/libgio-2.0.so.0
#11 0x0000fffff7e05704 in  () at /lib/aarch64-linux-gnu/libgio-2.0.so.0
#12 0x0000fffff7b879f0 in g_main_context_dispatch () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#13 0x0000fffff7b87d98 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#14 0x0000fffff7b880f0 in g_main_loop_run () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#15 0x0000aaaaaaaec528 in chip::DeviceLayer::GDBus_Thread() ()
    at ../../third_party/connectedhomeip/src/platform/Linux/PlatformManagerImpl.cpp:45
#16 0x0000aaaaaaaee9a4 in std::__invoke_impl<void, void (*)()>(std::__invoke_other, void (*&&)())
    (__f=@0xffffe8006cd8: 0xaaaaaaaec508 <chip::DeviceLayer::GDBus_Thread()>) at /usr/include/c++/10/bits/invoke.h:60
#17 0x0000aaaaaaaee930 in std::__invoke<void (*)()>(void (*&&)()) (__fn=@0xffffe8006cd8: 0xaaaaaaaec508 <chip::DeviceLayer::GDBus_Thread()>)
    at /usr/include/c++/10/bits/invoke.h:95
#18 0x0000aaaaaaaee8cc in std::thread::_Invoker<std::tuple<void (*)()> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) (this=0xffffe8006cd8)
    at /usr/include/c++/10/thread:264
#19 0x0000aaaaaaaee8a0 in std::thread::_Invoker<std::tuple<void (*)()> >::operator()() (this=0xffffe8006cd8)
    at /usr/include/c++/10/thread:271
#20 0x0000aaaaaaaee880 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)()> > >::_M_run() (this=0xffffe8006cd0)
    at /usr/include/c++/10/thread:215
--Type <RET> for more, q to quit, c to continue without paging--
#21 0x0000fffff7787e1c in  () at /lib/aarch64-linux-gnu/libstdc++.so.6
#22 0x0000fffff7f8ef74 in start_thread (arg=0x0) at pthread_create.c:463
#23 0x0000fffff76183dc in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78

mspang, 2020-11-01:
The same crashes happens after scanning QR code with the Android app.

erjiaqing, 2020-11-02:
Temporary workaround for this issue when using python device controller:

```
ble-scan-connect xxxxx
ble-disconnect
ble-scan-connect xxxxx
```

(After connect, run `ble-disconnect` and connect it again)

erjiaqing, 2020-11-02:
The cause of this issue is "Connect" property change event is not fired by bluez.

closed 2020-11-02 by rwalker-apple