project-chip/connectedhomeip #3555: Android app doesn't ask for permissions needed for BLE scan

mspang, 2020-11-01:
Installing the Android app trying to rendezvous via BLE fails with the following error

`11-01 06:50:58.905  1900  3456 E BluetoothUtils: Permission denial: Need ACCESS_FINE_LOCATION permission to get scan results`

markus-becker-tridonic-com, 2020-11-12:
@mspang I believe now the app has the permission: https://github.com/project-chip/connectedhomeip/blob/bd63c51d1de74850488face9c63384055e8e572f/src/android/CHIPTool/app/src/main/AndroidManifest.xml#L10

markus-becker-tridonic-com, 2021-02-12:
@vidhis88 i believe the app has the permission now and with 1e531e93610b19bf7143d3c6de46a8d3d12a13bd the BT adapter is enabled automatically. this issue can be closed, or?

vidhis88, 2021-02-12:
Agreed, can be closed.

mspang, 2021-02-16:
I don't think this is fixed?

02-16 17:15:04.001  1502  2468 E BluetoothUtils: Permission denial: Need ACCESS_FINE_LOCATION permission to get scan results

There needs to be a prompt right?

mspang, 2021-02-16:
AIUI <uses-permission> isn't a premission grant on any recent version of Android. We need to check the permission and prompt the user at runtime to grant it.

markus-becker-tridonic-com, 2021-02-17:
On my old phone, there was no requirement to check the runtime permissions.

Indeed ACCESS_FINE_LOCATION has `Protection level: dangerous` according to https://developer.android.com/reference/android/Manifest.permission#ACCESS_FINE_LOCATION and should be requested at runtime on newer Android versions.

Either the brief code:
```
if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.Q) {
    if (ContextCompat.checkSelfPermission(baseContext,
        Manifest.permission.ACCESS_FINE_LOCATION)
        != PackageManager.PERMISSION_GRANTED) {
            ActivityCompat.requestPermissions(
                this,
                arrayOf(Manifest.permission.ACCESS_FINE_LOCATION),
                    PERMISSION_CODE)
        }
}
```
or the more elaborate code from https://github.com/android/permissions-samples/blob/main/RuntimePermissionsBasicKotlin/Application/src/main/java/com/example/android/basicpermissions/MainActivity.kt could be used.

closed 2021-02-19 by andy31415