project-chip/connectedhomeip #3556: BLE advertisements are inconsistent between spec, commissioner apps, & examples

mspang, 2020-11-01:
Neither iOS nor Android apps will discover a device whose BLE advertisements match the current spec.

Currently they are not expecting a vendor/product id but the spec reserves space for one, and it looks like discriminator is not serialized correctly.

https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Darwin/BleConnectionDelegateImpl.mm#L128-L137

https://github.com/project-chip/connectedhomeip/blob/master/src/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/bluetooth/BluetoothManager.kt#L54-L57

The examples are also not consistent about which format they use, so for example Linux example cannot be discovered by mobile commissioners.

smides-nest, 2020-11-02:
@mspang - in a sidebar you mentioned that the current BLE Advertisement format is too short for iOS discovery but not for Android. Am I understanding that correctly? The above summary makes it sounds as if both iOS and Android both have issues with discovering the BLE Advertisement as currently specified here (https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/326).

mspang, 2020-11-02:
> @mspang - in a sidebar you mentioned that the current BLE Advertisement format is too short for iOS discovery but not for Android. Am I understanding that correctly? The above summary makes it sounds as if both iOS and Android both have issues with discovering the BLE Advertisement as currently specified here ([CHIP-Specifications/connectedhomeip-spec#326](https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/326)).

Updated the wording.

I was confused about Android, it's actually correct but there are two separate open PRs to change it back to the wrong format because of the inconsistencies.

closed 2020-11-02 by rwalker-apple

smides-nest, 2020-11-03:
@mspang -- Is "iOS requires a short length payload (4 bytes for VID/PID omitted)" fixed in b255818 #3557? So now iOS always expects the fully-specified Service Data and is able to process it?

PR #3557 (2020-11-02): Fix BLE advertisement service data format
 #### Problem

Several components are using an incorrect format for BLE advertisement.

 #### Summary of Changes
 
The correct format is implemented in ChipBLEDeviceIdentificationInfo.
Update the Linux, ESP32, Android, and Darwin ports to support the
correct format during Rendezvous.

Fixes #3556
commit b255818f (2020-11-02): Fix BLE advertisement service data format (#3557)

* Fix BLE advertisement service data format

Several components are using an incorrect format for BLE advertisement.

The correct format is implemented in ChipBLEDeviceIdentificationInfo.
Update the Linux, ESP32, Android, and Darwin ports to support the
correct format during Rendezvous.

Fixes #3556

* Drop matching VID/PID

They are optional and may be zero.

* Remove dead code

* Make wifi & thread BLE opcode consistent

* Use connected state rather than provision state