project-chip/connectedhomeip #3578: emberAfDoorLockClusterSetHolidayScheduleCallback needs to validate operatingModeDuringHoliday value

bzbarsky-apple, 2020-11-02:
#### Problem
Right now an invalid value of `operatingModeDuringHoliday` (out of range) will just get stored in the schedule table.  I would have expected a check here and an error return if it's out of range....

 #### Proposed Solution
Add a check.  I'm not sure how best to do that, given the enum is generated.  Maybe we could 
add something in the generated code that lets us check whether the value is valid for a given enum?

@vivien-apple

mrjerryjohns, 2022-01-31:
`emberAfDoorLockClusterSetHolidayScheduleCallback` is not implemented (still) in the SDK. Closing this as a stale issue for now since there isn't actually an illegal implementation yet.

closed 2022-01-31 by mrjerryjohns

bzbarsky-apple, 2022-02-01:
> emberAfDoorLockClusterSetHolidayScheduleCallback is not implemented (still)

More precisely, the implementation that was there (and buggy as described above) was removed, and a new one was not added.