project-chip/connectedhomeip #3580: Need to validate userType values in door lock server

bzbarsky-apple, 2020-11-02:
#### Problem
In `emberAfDoorLockClusterSetUserTypeCallback` and in `setUser` we have a `uint8_t userType` that we are casting to `EmberAfDoorLockUserType`, but nothing makes sure that it's a valid value.

 #### Proposed Solution
Add a check. I'm not sure how best to do that, given the enum is generated. Maybe we could
add something in the generated code that lets us check whether the value is valid for a given enum?

@vivien-apple

bzbarsky-apple, 2022-08-26:
This might not be relevant with the door lock cluster rewrites (and the fact that the "Set User Type" command is gone to start with in Matter)....

@mykrupp can you confirm that this is not an issue anymore, please?

mykrupp, 2022-08-26:
I don't see emberAfDoorLockClusterSetUserTypeCallback in the connectedhomeip repo anywhere. 

Although 4.26 in the doorlock spec indicates that Set User Type should be there. Was it decided at some point to remove the Set User Type command from matter?
[image]

bzbarsky-apple, 2022-08-27:
Set User Type is Zigbe-only.  The Matter equivalent is SetUser.  The question is whether SetUser validates the user type it's passed @mykrupp

mykrupp, 2022-08-28:
https://github.com/project-chip/connectedhomeip/pull/19813 This PR took care of it for SetUser

bzbarsky-apple, 2022-08-31:
Perfect!

closed 2022-08-31 by bzbarsky-apple