project-chip/connectedhomeip #3584: Consider switching various indexOrDestination bits to some sort of struct/union type

bzbarsky-apple, 2020-11-02:
#### Problem
There are a bunch of functions that take an `EmberOutgoingMessageType` and a numeric `indexOrDestination`.  The meaning of the indexOrDestination is then determined by the value of the `EmberOutgoingMessageType`.

 #### Proposed Solution
It might be better to have a struct that has an EmberOutgoingMessageType member and then a union of whatever things `indexOrDestination` is supposed to be able to represent, with accessors to get at various information.

I _think_ `indexOrDestination` can be a `NodeId` or a `GroupId` or an index into the binding table.  @selissia does that sound about right?

selissia, 2020-11-03:
> I _think_ `indexOrDestination` can be a `NodeId` or a `GroupId` or an index into the binding table. @selissia does that sound about right?

Yes, that's right. We discussed this on the #chipsilabsintegration a while ago (for some reason I can no longer see the old messages). 

￼

bzbarsky-apple, 2020-11-07:
> for some reason I can no longer see the old messages

Because we're not paying Slack (enough?) so they only keep a certain number of messages.

kghost, 2020-11-18:
I'm working on this, this will be a subtask of integrate ember with exchange layer.

I'll add a `EmberContext` to capture all short live variables, and it will replace `indexOrDestination`. The binding and the destination node id will be the member of `EmberContext`

bzbarsky-apple, 2020-11-18:
Please don't use "Ember" in new names; we are generally aiming to eliminate that naming convention from the code, but it's waiting on us having control of the generator for the generated files so we can change those in sync with the non-generated ones.

Also, "Context" is kind of generic.... the right thing here is "MessageDestination" or the like, right?

kghost, 2020-11-18:
Let's continue the discussion in the linked PR.

woody-apple, 2021-01-29:
@kghost is this still required? Please re-open if needed! (Linked PR is closed)

bzbarsky-apple, 2021-01-29:
Yes, this is still needed.

kghost, 2021-02-02:
This is block by #4019, see https://github.com/project-chip/connectedhomeip/issues/4170#issuecomment-771352570

closed 2021-05-05 by bzbarsky-apple