project-chip/connectedhomeip #3593: Chip tool iOS app doesn't see the BT advertisement from the raspi device

cjandhyala, 2020-11-03:
Running the "chip-tool-server" executable built for lightApp on raspi with Ubuntu Desktop OS, sends the BT advertisements. Generated the QR code with the string from the log and used the CHIP TOOL iOS app to look for the BT advertisements. CHIP TOOL doesn't seem to catch the advertisements.

sudo ./chip-tool-server 
[sudo] password for pi: 
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*K2H5+B9SS0]
CHIP:IN: New pairing for key 0!!
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bus acquired for name CHIP-1383
CHIP:DL: CREATE service object at /chipoble/e5e5/service
CHIP:DL: Create characteristic object at /chipoble/e5e5/service/c1
CHIP:DL: Create characteristic object at /chipoble/e5e5/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/e5e5/service
CHIP:DL: CHIP BTP C2 /chipoble/e5e5/service
CHIP:DL: wpa_supplicant: connected to wpa_supplicant proxy
CHIP:DL: wpa_supplicant: WiFi interface: /fi/w1/wpa_supplicant1/Interfaces/0
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: wpa_supplicant: connected to wpa_supplicant interface proxy
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1001 ms
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent , 16387
CHIP:DL: Create adv object at /chipoble/e5e5/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x67, 0x05, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent , 16388
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent , 16389

woody-apple, 2020-11-03:
This appears to work for ESP32, nRF, and EFR. Probably should investigate at the linux platform side first.

woody-apple, 2021-01-29:
Duplicate of: https://github.com/project-chip/connectedhomeip/pull/4542

closed 2021-01-29 by woody-apple