project-chip/connectedhomeip #3627: ZCL Cleanup Plugin Macro

jepenven-silabs, 2020-11-03:
#### Problem
Since plugins macros are not generated anymore by ZAP they were moved to `/src/app/util/plugin-config.h` However no cleanup was done to see which one we should keep.

 #### Proposed Solution
Removed unused macro from `/src/app/util/plugin-config.h`

vivien-apple, 2020-11-06:
I have gone thought all the macros. Here is the list of macros that are currently used from the set of macros from `plugin-config.h` from #3464 :

```
#define EMBER_BINDING_TABLE_SIZE 10
#define EMBER_AF_PLUGIN_COLOR_CONTROL_SERVER_XY
#define EMBER_AF_PLUGIN_COLOR_CONTROL_SERVER_TEMP
#define EMBER_AF_PLUGIN_COLOR_CONTROL_SERVER_HSV
#define EMBER_AF_PLUGIN_GROUPS_SERVER 
#define EMBER_AF_PLUGIN_IAS_ZONE_CLIENT
#define EMBER_AF_PLUGIN_IAS_ZONE_CLIENT_MAX_DEVICES 10
#define EMBER_AF_PLUGIN_IAS_ZONE_SERVER_ZONE_TYPE 541
#define EMBER_AF_PLUGIN_LEVEL_CONTROL
#define EMBER_AF_PLUGIN_LEVEL_CONTROL_MAXIMUM_LEVEL 255
#define EMBER_AF_PLUGIN_LEVEL_CONTROL_MINIMUM_LEVEL 0
#define EMBER_AF_PLUGIN_LEVEL_CONTROL_RATE 0
#define EMBER_AF_PLUGIN_REPORTING
#define EMBER_AF_PLUGIN_REPORTING_TABLE_SIZE 5
#define EMBER_AF_PLUGIN_REPORTING_ENABLE_GROUP_BOUND_REPORTS
#define EMBER_AF_PLUGIN_SCENES
#define EMBER_AF_PLUGIN_SCENES_TABLE_SIZE 3
```

vivien-apple, 2020-11-06:
That said I don't think adding `src/app/util/plugin-config.h` is the way to go.

I would have expect ZAP to have generated something  in `gen_config.h`.

For example, if the groups server is enabled, it could have generated: 
```
#define EMBER_AF_PLUGIN_GROUPS_SERVER
```

Of if any of the attributes is tagged as reportable, it could have generated:
```
#define EMBER_AF_PLUGIN_REPORTING
```

Or if the Level Control server has been selected, it could have generated:
```
#define EMBER_AF_PLUGIN_LEVEL_CONTROL
#define EMBER_AF_PLUGIN_LEVEL_CONTROL_MAXIMUM_LEVEL 255
#define EMBER_AF_PLUGIN_LEVEL_CONTROL_MINIMUM_LEVEL 0
#define EMBER_AF_PLUGIN_LEVEL_CONTROL_RATE 0
```

Or depending on which commands have been selected for the Color Control cluster it may have generated:
```
#define EMBER_AF_PLUGIN_COLOR_CONTROL_SERVER_XY
```
but not
```
#define EMBER_AF_PLUGIN_COLOR_CONTROL_SERVER_TEMP
```

I understand that the generated templates may differs for a different framework than CHIP, but since in this case we are speaking of templates targeting CHIP itself I would have expected those to match the current code architecture.

So I think that's something we need to fix in the current templates that will land in #3638 since having those `#define` be part of `src/app/` have some side effects on the other clusters.

PR #3872 (2020-11-17): Add '#define EMBER_AF_PLUGIN_{{clusterName}}' and related to src/app/…
…zap-templates/gen_config.h

 #### Problem

#3464 get rids of most of the `#define`  from `gen/gen_config.h`. That's a nice cleanup but some of them are still required in order to build properly. Notably the one that are related to `Plugins`.

I have not found a good way to define `EMBER_AF_PLUGIN_REPORTING` conditionaly. I have opened #3871 for it.

 #### Summary of Changes
 * Add `EMBER_AF_PLUGIN_*` macro generations to `src/app/zap-templates/gen_config.zapt`

Fixes #3627
closed 2020-11-17 by rwalker-apple