project-chip/connectedhomeip #3634: IP address not assigned on Wlan0 interface on raspi

cjandhyala, 2020-11-03:
IP address not assigned on Wlan0 interface on raspi after wifi is provisioned.

yufengwangca, 2020-11-04:
@cjandhyala CHIP does not has the API to assign the IP address right now. ProvisionWiFi only associate the device to the specified BSS via its SSID

cjandhyala, 2020-11-04:
From the chip Device Controller ,  I see Rendezvoud Complete

chip-device-ctrl > connect 12345678
CHIP: [BLE] subscribe complete, ep = 0x106d3a208
CHIP: [BLE] peripheral chose BTP version 3; central expected between 2 and 3
CHIP: [BLE] using BTP fragment sizes rx 20 / tx 128.
CHIP: [BLE] local and remote recv window size = 3
CHIP: [CTL] Remote device completed SPAKE2+ handshake

CHIP: [CTL] Will Call on mOnNewConnection(0x101aa5de0)

Rendezvoud Complete
CHIP: [NP] Sending Network Creds. Delegate 0x7fa885a67018

Connected
chip-device-ctrl > 
----------------------------------------
on the raspi Device, its connected to new SSID. But the Wlan0 interface is not getting any IP address assigned.  

CHIP:BLE: Indication confirmation, 0xffff7c02a380
CHIP:DL: HandlePlatformSpecificBLEEvent , 7
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffff7c02a380
CHIP:NP: Received kWiFiAssociationRequest.

CHIP:NP: LinuxNetworkProvisioningDelegate: SSID: Partyguys
CHIP:DL: wpa_supplicant: added network: SSID: Partyguys: /fi/w1/wpa_supplicant1/Interfaces/0/Networks/1
CHIP:DL: wpa_supplicant: connected to network: SSID: Partyguys
CHIP:SVR: Device was assigned network credentials
CHIP:DL: HandlePlatformSpecificBLEEvent , 6
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffff7c02a380
CHIP:DL: HandlePlatformSpecificBLEEvent , 6
CHIP:BLE: Indication confirmation, 0xffff7c02a380
CHIP:DL: HandlePlatformSpecificBLEEvent , 7
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:BLE: Write request received debug 0xffff7c02a380
CHIP:DL: HandlePlatformSpecificBLEEvent , 6
CHIP:BLE: Indication confirmation, 0xffff7c02a380

yufengwangca, 2020-11-04:
The logic of getting IP address should be reside in app, CHIP does not handle it. Which App do you use? chip-device-ctrl? Does it have this logic?

erjiaqing, 2020-11-04:
Run dhcpcd on wlan0

WiFi on Linux platforms are controlled by direct calling wpa_supplicant dbus API. Which will not run dhclient on wlan0 interface, so you should manaully run this to get an IP address. You can also add it to service and run it on boot.

  $ sudo dhclient wlan0
Test the device using ChipDeviceController on your laptop / workstation etc.

yufengwangca, 2020-11-04:
You can run sudo dhclient wlan0 or use ifconfig on wlan0, I guess he means how to get the ip address without intervention

erjiaqing, 2020-11-04:
> You can run sudo dhclient wlan0 or use ifconfig on wlan0, I guess he means how to get the ip address without intervention

Can be done by adding dhclient as a service when system boot up.

yufengwangca, 2020-11-04:
@erjiaqing There is a C-API to retrieve the IP address, but we first need to decide if we need to add this API in CHIP. From my perspective, this is outside of CHIP

yufengwangca, 2020-11-04:
> Run dhcpcd on wlan0
> 
> WiFi on Linux platforms are controlled by direct calling wpa_supplicant dbus API. Which will not run dhclient on wlan0 interface, so you should manaully run this to get an IP address. You can also add it to service and run it on boot.

"Add to service" might not help, we need to get the IP address right after the device get associated to AP

cjandhyala, 2021-01-29:
running this cmd 'sudo dhclient wlan0' as service at system boot may not help as the device is not associate to an AP. Need to run this as part of the app after the device connected to an AP. any comments

yufengwangca, 2021-02-02:
This should be handled by the app, after the device get provisioned by app, the app need to run dhcp to request the IP address.

closed 2021-03-03 by andy31415