project-chip/connectedhomeip #3639: Embedded mDNS subscriber

gjc13, 2020-11-04:
#### Problem

mDNS subscription support is poor in embedded systems. Zephyr and lwip doesn't support browsing and resolving services and ESP forces a timeout that will stall the event loop.

 #### Proposed Solution
Implement an embedded mDNS library inside CHIP repo.

Damian-Nordic, 2021-03-31:
@gjc13 Is there anything still needed for this issue? There exists SRP-based implementation for Thread devices, avahi, ESP32 and minimal mdns for Wi-Fi devices.

gjc13, 2021-04-01:
There were discussions about whether we need to map the node id to the IP addresses on end devices. I believe it's not a must to do so for end devices now. @yunhanw-google @kghost Is it still required?

kghost, 2021-04-06:
I think it is still on schedule. @erjiaqing may have gotten more context.

Damian-Nordic, 2021-04-13:
Note that we have already implemented the `Resolver` interface based on Linux (avahi), Darwin. Also, the minimal mDNS  implementation is pending (PR created by Andy: https://github.com/project-chip/connectedhomeip/pull/5907). The interface is then used in the DeviceController to update IP addresses in the internal structures (see https://github.com/project-chip/connectedhomeip/commit/e047d27d384e8243e19ccdcf85d5389eb45c10ac). Hence, I think the task may be too general and needs clarification.

franck-apple, 2021-04-15:
Accessory to accessory is not needed for TE2