project-chip/connectedhomeip #3663: Unbounded stack in payloadBase41RepresentationWithTLV()

kpschoedel, 2020-11-04:
#### Problem

payloadBase41RepresentationWithTLV() allocates stack space based on an argument, with no upper bound. This could lead to stack exhaustion on small devices.

 #### Proposed Solution

TBD. The caller's callers pass in a destination buffer, which might be usable as temporary space.

Code in this path also uses heap-allocated std::strings, which should ultimately be removed for issue #2748 (All Embedded Device-side Code Needs to Be Scrubbed of Direct Heap Usage).

kpschoedel, 2020-11-06:
decimalStringWithPadding() in src/setup_payload/ManualSetupPayloadGenerator.cpp added to this issue.

Per https://github.com/project-chip/connectedhomeip/pull/3683/files#r518644463 these uses are probably OK, and just need to be recorded as such.

alexhqwang, 2021-07-16:
Not sure how to change assignee, but I am looking into this since the Nest RTOS thermostat does not support std::string.

alexhqwang, 2021-07-27:
PR has merged, this can be closed. thanks!

closed 2021-07-27 by kpschoedel