project-chip/connectedhomeip #3687: nRF Connect documented flow uses external docker image

mspang, 2020-11-05:
Currently nRF platform builds use connectedhomeip/chip-build-nrf-platform and the README suggests using nordicsemi/nrfconnect-chip .

This breaks the assumption that all CHIP images derive from chip-build, so that image no longer provides a central place to install dependencies or run preliminary commands on Linux.

Let's contribute the improvements in nordicsemi/nrfconnect-chip back to CHIP instead of forking the image.

mspang, 2020-11-05:
@Damian-Nordic @kkasperczyk-no

mspang, 2020-11-05:
@rwalker-apple

woody-apple, 2021-01-29:
@Damian-Nordic Could someone at nordic help out with this? It seems like we'd want a smooth Nordic flow for developers!

kkasperczyk-no, 2021-01-29:
@woody-apple are there any problems with starting development on Nordic platform? Since that issue was raised we significantly improved our docs and created multiple user guides to simplify development, so we were convinced that currently working with our platform is much easier. If you know about any difficulties I would appreciate your feedback to help us solve it.

I guess this issue was more related to some CHIP workflow incompatibilities than appearing problems when working with our platform, however we somehow lost it from our sight. We will take a look on that and try to find differences between current versions of nordicsemi/nrfconnect-chip and connectedhomeip/chip-build-nrf-platform images.

closed 2022-10-09 by stale[bot]