project-chip/connectedhomeip #3689: Fix assert() misuse in data model code

bzbarsky-apple, 2020-11-05:
#### Problem
We have uses of `assert()` in data model code where functions with side-effects are part of the `assert()` argument.  This does not play well with `-DNDEBUG`

 #### Proposed Solution
Fix `assert()` misuse.

That said, it's not clear to me whether the resulting code can end up doing unsafe things if `NDEBUG` due to pressing on past the asserts.....  @selissia

selissia, 2020-11-05:
> That said, it's not clear to me whether the resulting code can end up doing unsafe things if NDEBUG due to pressing on past the asserts..... @selissia
Sorry, could you please clarify the above sentence?

bzbarsky-apple, 2020-11-05:
@selissia So an example of code I am looking at is:
```
static uint8_t readColorMode(uint8_t endpoint)
{
    uint8_t colorMode;

    assert(EMBER_ZCL_STATUS_SUCCESS ==
           emberAfReadServerAttribute(endpoint, ZCL_COLOR_CONTROL_CLUSTER_ID, ZCL_COLOR_CONTROL_COLOR_MODE_ATTRIBUTE_ID,
                                      (uint8_t *) &colorMode, sizeof(uint8_t)));

    return colorMode;
}
```
in color-control-server.  I'd want to rewrite this as:
```
static uint8_t readColorMode(uint8_t endpoint)
{
    uint8_t colorMode;

#ifndef NDEBUG
    EmberAfStatus status =
#endif // NDEBUG
        emberAfReadServerAttribute(endpoint, ZCL_COLOR_CONTROL_CLUSTER_ID, ZCL_COLOR_CONTROL_COLOR_MODE_ATTRIBUTE_ID,
                                   (uint8_t *) &colorMode, sizeof(uint8_t));
    assert(status == EMBER_ZCL_STATUS_SUCCESS);

    return colorMode;
}
```
but can we then ever reach the `return colorMode;` bit in practice?
```

selissia, 2020-11-05:
Old code:
With NDEBUG defined: Return uninitialized colorMode. (In general case: skip execution of some function with side effects)
With NDEBUG not defined: In case of error abort(), otherwise return valid colorMode

New code:
With NDEBUG defined: Return potentially invalid colorMode. (In general case: guarantee execution of some function with side effects)
With NDEBUG not defined:In case of error abort(), otherwise return valid colorMode

I'd say that the outcome is better with the new code. If some old logic relied on NDEBUG being enabled then good riddance, better catch its effect now than later.

bzbarsky-apple, 2020-11-05:
That's fair.  The bad case in the new code is NDEBUG defined and an error happens, but the old code was also bad in that situation.

I guess I'm trying to figure out whether I should do more extensive chnages to make sure we handle errors more gracefully when NDEBUG is defined or whether these errors are pretty theoretical (in that they can only happen if the attributes are not present on the endpoint, and for this code they would have to be).

closed 2020-11-09 by bzbarsky-apple