project-chip/connectedhomeip #3718: Investigate if `-ffunction-sections`helps with linker issues over Linux

vivien-apple, 2020-11-06:
#### Problem

In #3678 and #2253 some workarounds has been done (like adding some symbols / removing some symbols) in order to make the linker happy.

Instead of doing that all the time, it worth checking if adding `-ffunction-sections` to cflags and c++ flags fixes the linker issues (from https://github.com/project-chip/connectedhomeip/pull/3678#discussion_r518936785)

vivien-apple, 2020-11-07:
It seems like `-ffunction-sections` is already the default option: https://github.com/project-chip/connectedhomeip/blob/master/build/config/compiler/BUILD.gn#L268

This may answer the question from #1223. I believe this bug or #1223 can be closed as a duplicate.

The linking errors disappears if you add `-Wl,--gc-sections` to the `flags`sections of `optimise_zero` (https://github.com/project-chip/connectedhomeip/blob/master/build/config/compiler/BUILD.gn#L110).

`optimize_zero` is the debug build options for OSes that are not real time. So one other option could be to not do debug build for cirque.

bzbarsky-apple, 2020-11-08:
I'd argue we should add `-Wl,--gc-sections` for debug builds, personally.  I guess the tradeoff is that one might want functions that only get called from the debugger?

@mspang @andy31415

mspang, 2020-11-08:
We shouldn't have unresolved symbols. Debug build is arguably diagnosing a bug here, which is its purpose.