project-chip/connectedhomeip #3722: Are you considering supporting some devices which don't support IP?

rise0chen, 2020-11-08:
Are you considering supporting some devices which don't support IP?

gerickson, 2020-11-09:
No, the focus of Project Connected Home over IP is IP-based IoT devices. There may be defined opportunities for bridging and proxies to legacy solutions and technologies.

rise0chen, 2020-11-09:
It will be a very big constraints for this project.
A lot of devices use ZigBee, BLE,IR and CAN,485 Bus. And,lots of communication can't support IP Stack.

gerickson, 2020-11-09:
That's OK; the goal and focus of Project CHIP is IP. There remains room in the industry for those interested in focusing on non-IP solutions.

maxCGN, 2020-11-15:
Yeah so basically all will be the same.

Hedda, 2021-01-28:
> It will be a very big constraints for this project.
> A lot of devices use ZigBee, BLE,IR and CAN,485 Bus. And,lots of communication can't support IP Stack.

Solution for this will probably be bridge/gateway appliances and software which acts like a proxy server at the application level.

That is, you make a Zigbee/BLE/Z-Wave to CHIP bridge appliance/software products which convert those to Connected Home IP.

Those would basically work the same as all existing bridges/gateways (a.k.a. hubs) that need for Zigbee/BLE/Z-Wave, etc. today.

woody-apple, 2021-01-29:
Closing based on comments from @gerickson & @Hedda, please re-open if you have further questions

closed 2021-01-29 by woody-apple