project-chip/connectedhomeip #3736: Consider eliminating strncpy from our code

bzbarsky-apple, 2020-11-10:
#### Problem
`strncpy` will sometimes null-terminate and sometimes not.

 #### Proposed Solution
Use `strlcpy` instead.

@rwalker-apple @pan-apple @mspang

mspang, 2020-11-10:
Linux's libc doesn't have `strlcpy`.

bzbarsky-apple, 2020-11-10:
Ah, that's unfortunate.  I guess not much we can do here, then, apart from making sure all `strncpy` callers properly check return values...

closed 2020-11-10 by bzbarsky-apple

rwalker-apple, 2020-11-11:
strlcpy() isn't terribly hard to implement

gerickson, 2020-11-13:
Agreed. I know we've moved well-past autotools; however, in typical autotools projects, you have a compat directory and where implementations of things like strlcpy aren't natively available, something from compat is used. In fact, I suspect we could "third_party" an existing `compat` implementation from another project with a suitable license and call it done.

rwalker-apple, 2020-11-16:
> Linux's libc doesn't have `strlcpy`.

libbsd-dev has strlcpy()...