project-chip/connectedhomeip #3764: Need new type for TLV tag that were not a raw integer; that would make all this enforceable by the type system.

yunhanw-google, 2020-11-11:
This needs to document the arguments and general functionality.  In particular, at least the following should be documented:

1. What is this assuming about the state of the reader when this function is called?  Presumably that it's positioned right before the first element inside a container?  Or is it meant to be used more generically?  (I know it will do something sane no matter what element it's positioned at; that's an implementation detail and we should document what the intended API use case is.)
2. What does `tagInApiForm` mean?  As a consumer, how do I figure out what to pass in here for a given tag value defined by some TLV-based specification?  Perhaps document that this must be the sort of thing returned by a `ContextTag()` or `CommonTag()` or `ProfileTag()` call?  I wish we had a type for "api tag" that were not a raw integer; that would make all this enforceable by the type system.  Maybe file a followup issue to do that?
3. Is the fact that `destReader` is not modified on failure happenstance or part of the API?

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/3695#discussion_r519096653_

holbrookt, 2022-02-01:
@yunhanw-google this issue is very old so I'm removing the 1.0 tag. Please re-label if this is wrong

bzbarsky-apple, 2022-02-01:
Fixed in https://github.com/project-chip/connectedhomeip/pull/13219

closed 2022-02-01 by bzbarsky-apple