project-chip/connectedhomeip #3785: ESP32 NimBLE: Not able to complete Rendezvous session using chip-tool

prasad-alatkar, 2020-11-11:
#### Problem

 * expected behavior:
   - Should be able to connect to ESP32 using chip-tool and complete Rendezvous session.
 * actual behavior
   - Even after running `./out/debug/chip-tool echo ble 12345678 3840` on mac not able to connect to ESP32(NimBLE). This behavior has been seen after `Fix BLE advertisement service data format (#3557)` commit. 
 * steps to reproduce
   - Run `all-clusters-app` app on ESP32 with NimBLE BLE stack.
   - On mac, run `./out/debug/chip-tool echo ble 12345678 3840` , it won't connect to ESP32
 * system configuration
  - Enable NimBLE stack (enabled by default)
  - Choose Rendezvous Mode as BLE.

 #### Proposed Solution

If we comment out [kFlag_AdvertisingRefreshNeeded](https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/nimble/BLEManagerImpl.cpp#L276) the chip-tool can connect to ESP32.

prasad-alatkar, 2020-11-18:
Update: This issue occurs when I provide SSID + password through static configuration(`sdkconfig`) . Since the SSID and credentials are provided to device, I get `kWiFiConnectivityChange` platform event which retriggers the device advertisement and then chip-tool is unable to connect to ESP32. Is this expected behavior ? CC: @mspang @dhrishi

dhrishi, 2021-04-07:
Hey @woody-apple I think this isn't relevant now. Should we close this?

dhrishi, 2021-05-25:
Closing as this is not relevant now.

closed 2021-05-25 by dhrishi