project-chip/connectedhomeip #3793: Investigate FLASH and RAM usage of ESP32 (vanilla ESP-IDF)

dhrishi, 2020-11-12:
#### Problem
We need to optimize on the FLASH and RAM usage on the ESP32

 #### Proposed Solution
For that, we would first need the numbers with vanilla ESP-IDF and the required functionality through a sample app (incrementally -- so that the numbers could be tracked incrementally as well). Once that is done, we can further try to optimize it through the appropriate configuration options.

After that we will have a baseline and then we can use the similar software setup in the CHIP demo with ESP32 and investigate its usage.

dhrishi, 2020-11-12:
Using the release/v4.1 branch of ESP-IDF, below are the numbers for the default Wi-Fi station example:

1. Default Wi-Fi station app - Free heap (after connect): 226748
$ idf_size.py build/wifi_station.map
```
Total sizes:
 DRAM .data size:   18980 bytes
 DRAM .bss  size:   23984 bytes
Used static DRAM:   42964 bytes ( 137772 available, 23.8% used)
Used static IRAM:   89099 bytes (  41973 available, 68.0% used)
      Flash code:  466103 bytes
    Flash rodata:   98308 bytes
Total image size:~ 672490 bytes (.bin may be padded larger)
```
`Free heap after Wi-Fi connection: 226748 bytes`

2. Previous + Code size optimization + Optimized configuration values for Wi-Fi and lwIP
$ idf_size.py build/wifi_station.map
```
Total sizes:
 DRAM .data size:   18360 bytes
 DRAM .bss  size:   23984 bytes
Used static DRAM:   42344 bytes ( 138392 available, 23.4% used)
Used static IRAM:   94451 bytes (  36621 available, 72.1% used)
      Flash code:  423927 bytes
    Flash rodata:   93732 bytes
Total image size:~ 630470 bytes (.bin may be padded larger)
```
`Free heap after Wi-Fi connection: 237080 bytes`

3. Previous + Bluetooth controller (1 max connection) + No host + release classic BT memory
$ idf_size.py build/wifi_station.map
```
Total sizes:
 DRAM .data size:   20164 bytes
 DRAM .bss  size:   26408 bytes
Used static DRAM:   46572 bytes (  78008 available, 37.4% used)
Used static IRAM:  125823 bytes (   5249 available, 96.0% used)
      Flash code:  495663 bytes
    Flash rodata:   99060 bytes
Total image size:~ 740710 bytes (.bin may be padded larger)
```
`Free heap after Wi-Fi connection: 195936 bytes`

4. Previous + NimBLE host (1 max connection) Peripheral role
$ idf_size.py build/wifi_station.map
```
Total sizes:
 DRAM .data size:   20492 bytes
 DRAM .bss  size:   31696 bytes
Used static DRAM:   52188 bytes (  72392 available, 41.9% used)
Used static IRAM:  126019 bytes (   5053 available, 96.1% used)
      Flash code:  559623 bytes
    Flash rodata:  107148 bytes
Total image size:~ 813282 bytes (.bin may be padded larger)
```
`Free heap after Wi-Fi connection: 174348 bytes`

5. Previous + Disable logs completely 
$ idf_size.py build/wifi_station.map
```
Total sizes:
 DRAM .data size:   16004 bytes
 DRAM .bss  size:   31696 bytes
Used static DRAM:   47700 bytes (  76880 available, 38.3% used)
Used static IRAM:  125367 bytes (   5705 available, 95.6% used)
      Flash code:  549579 bytes
    Flash rodata:   91876 bytes
Total image size:~ 782826 bytes (.bin may be padded larger)
```
`Free heap after Wi-Fi connection: 178872 bytes`

6. Previous + Disable asserts
$ idf_size.py build/wifi_station.map
```
Total sizes:
 DRAM .data size:   15468 bytes
 DRAM .bss  size:   31688 bytes
Used static DRAM:   47156 bytes (  77424 available, 37.9% used)
Used static IRAM:  121871 bytes (   9201 available, 93.0% used)
      Flash code:  537755 bytes
    Flash rodata:   59764 bytes
Total image size:~ 734858 bytes (.bin may be padded larger)
```
`Free heap after Wi-Fi connection: 179540 bytes`

woody-apple, 2021-02-02:
@dhrishi Are there any next steps here?

dhrishi, 2021-02-04:
Hey @woody-apple  The updated configuration options mentioned here would be applied to the temperature-measurement app (https://github.com/project-chip/connectedhomeip/issues/3977) Then the functionality will be tested and the optimized footprint verified. @sweetymhaiske  Can you please take a look and update?

dhrishi, 2021-02-09:
Closing this as the ESP-IDF analysis is done. Fix for #3977 will now add these changes to CHIP repo

closed 2021-02-09 by dhrishi