project-chip/connectedhomeip #3822: [nrfconnect] Remove temporary timeval problem fix after merging solution from Zephyr.

kkasperczyk-no, 2020-11-13:
#### Problem
Temporary solution was introduced in order to fix timeval type size problem resulting in improper operation of the select method. It's an internal Zephyr implementation problem and it will be fixed there.

 #### Proposed Solution
Remove changes introduced in https://github.com/project-chip/connectedhomeip/issues/3799 after merging fix from Zephyr project.

woody-apple, 2021-02-02:
@kkasperczyk-no has this been done yet? any ETA?

kkasperczyk-no, 2021-02-03:
@woody-apple It can't be removed yet. Fix was merged to Zephyr master quite long ago, but in CHIP we got Zephyr revision associated with the nRF Connect SDK version and currently recommended 1.4.0 version does not have it. New version 1.5 of nRF Connect SDK should be released in next few weeks and after that we will update it in CHIP. I remember about that issue, but thank you for reminding and tracking issues status!

PR #5103 (2021-03-03): [nrfconnect] Added changes related with the nRF Connect SDK update to 1.5.0.
 #### Problem
nRF Connect SDK 1.5.0 version was released, so CHIP should be updated
to be able benefit from it.

 #### Summary of Changes
* Added project name to the zephyr/module.yml to distinguish upstream CHIP project, from the one pulled by the nRF Connect SDK.
* Removed temporary timeval fix, as it already exist in Zephyr.
* Added setting MBEDTLS_PK_C and MBEDTLS_PK_C_WRITE in config file.
* Aligned examples readmes.

 Fixes #3822
closed 2021-03-03 by andy31415