project-chip/connectedhomeip #3829: [darwin] CHIPTool supports in service discovery only full 128 B BLE service UUID format.

kkasperczyk-no, 2020-11-13:
#### Problem
In the didDiscoverServices method there is a service.UUID.data isEqualToData:chipServiceUUID.data condition, which is fulfilled only in case of using 128 B size UUID and fails for the short 16 UUID format. I think it would be good to support both size formats, so I would like to ask if it's intentional operation or it should be expanded about short UUID format support?

 #### Proposed Solution
If there are not any objections, add support for short 16 B UUID format in the service discovery.

kkasperczyk-no, 2020-12-09:
@shana-apple are you working on this? I saw that #4098 was merged, so actually my problem was solved. I wrote about supporting both 16 and 128 bit UUID formats, because I thought that there are some reasons to use full format on the ESP32, but if all samples were aligned to use 16 bit I would like to close this issue.

shana-apple, 2020-12-09:
> @shana-apple are you working on this? I saw that #4098 was merged, so actually my problem was solved. I wrote about supporting both 16 and 128 bit UUID formats, because I thought that there are some [...]

No I am not yet so feel free to close if it is not applicable anymore.

closed 2020-12-09 by kkasperczyk-no