project-chip/connectedhomeip #3830: [darwin] Add Thread provisioning for CHIPTool.

kkasperczyk-no, 2020-11-13:
#### Problem
Currently CHIPTool iOS version supports only Wi-Fi network provisioning and in order to enable also testing with the Thread devices it is necessary to add support for Thread provisioning.

 #### Proposed Solution
Add support for provisioning Thread network credentials, basing on the Android CHIPTool solution (https://github.com/project-chip/connectedhomeip/pull/3390).

erjiaqing, 2020-12-03:
Since we will remove network provsioning from rendezvous session and move to ZCL clusters, I will work on https://github.com/project-chip/connectedhomeip/issues/3945 to add network provisoning cluster to chip, and then implement thread network provisioning functions.

erjiaqing, 2021-01-21:
Replace by #4373

closed 2021-01-21 by erjiaqing