project-chip/connectedhomeip #3840: Move common parts example apps to <app>-common

tima-q, 2020-11-16:
#### Problem
Current example apps share a lot of code for the top-level - platform-agnostic - behavior
Moving forward maintenance will become harder, code is already diverging on some points (fi QRCode printing)

 #### Proposed Solution
Moving common parts to the common code directory of the example.
Some parts, recurring in all examples, could even be globally common for all example applications - for example LEDWidget

Main differences seen so far in main startup/timer use.

Identified communalities:
* Lock-App
** BoltLockManager.cpp
** ZclCallbacks.cpp
** AppTask.cpp

* Lighting-app
** LightingManager.cpp
** ZclCallbacks.cpp
** AppTask.cpp

* Overall
** LEDWidget.cpp
** ButtonHandler.cpp

tima-q, 2020-11-17:
Before starting on this issue, feedback is required from other example maintainers on feasibility and alignment
This change will simplify maintenance, but also align content of the applications. 
Feature differentiation might be slightly more difficult.

From the contributors:
@Damian-Nordic 
@jmartinez-silabs , @jepenven-silabs 
@doru91

jmartinez-silabs, 2020-11-18:
Hi,
It can be done but we would need absraction code for the gpio, led api, timers, etc. 
We use FreeRTOS timers and I think nordic use something else now with NRFConnect. I am not sure for other manufacturer.
There are also hardware considerations. For example, nordic Dev kits have 4 push buttons and 4 LEDs they can use while ours(silabs dev kit) only have two. We have an onboard LCD for thr QR code and nordic use an external one. 

Personally, I think it is ok to have code duplicated for some of this because it's example code and not Chip application related. Also It is easier for outside users to clone the repo and use the example for the dev kit they have and navigate the code and API relative to the manufacturer platform they use without an abstraction layer on top.

For the ZclCallbacks.cpp that should be straight foward and easy to move to common lock-app/common lighting-app

woody-apple, 2021-02-02:
@vivien-apple thoughts about this given the tree now?

vivien-apple, 2021-02-05:
> @vivien-apple thoughts about this given the tree now?

I guess this issue is mostly, out of my scope. It is related to sharing more code into some examples that are really application specific.
Personally, based on the answer from @jmartinez-silabs I would say that it can be closed  at the moment. It can be reopened if someone really believe this is important.

closed 2021-02-05 by vivien-apple