project-chip/connectedhomeip #3871: Conditionaly define EMBER_AF_PLUGIN_REPORTING

vivien-apple, 2020-11-17:
#### Problem

With the move to #3464, `EMBER_AF_PLUGIN_REPORTING` is enabled globally (see #3608)

It should be enable only if needed.

 #### Proposed Solution

One possible solution would be to fix the templates helper and add an `hasReportableAttributes` method that will conditionally include this macro in `gen_config.zapt`

One other possible solution would be for the `gen_config.zapt` #define to be part of a global header that is generated at built-time. It may also enable the ability to automatically include the right dependencies for building depending on the set of choosen cluster.

bzbarsky-apple, 2020-11-17:
I believe in practice in CHIP all attributes are reportable, and all nodes have attributes that can change (e.g. in Basic), so reporting is always needed...

woody-apple, 2021-02-02:
Per @bzbarsky-apple 's comment above, closing, makes sense that reporting is always required. Please re-open if we're missing something!

closed 2021-02-02 by woody-apple