project-chip/connectedhomeip #3945: Add network provisioning cluster xml

yunhanw-google, 2020-11-20:
#### Problem
per spec https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/309, we need initial network provisioning cluster xml

erjiaqing, 2020-12-08:
The code for this issue should be submitted to zap repo. Will work on this since related spec is ready.

franck-apple, 2021-02-26:
Re-opening. The issue should be closed once it’s working end to end.

erjiaqing, 2021-02-26:
> Re-opening. The issue should be closed once it’s working end to end.

Just submitted a PR for basic support (only supports WiFi, and only on lighting app)

Will create seperate PR for scale the network commissioning to all example apps and for details functions. (Or maybe add a list to this PR? However, I'd like to create a project for all commissioning works.)

Note: Some features depends on other features in CHIP(like fabric commissioning) to be finished first.

closed 2021-03-02 by andy31415