project-chip/connectedhomeip #3957: Fix unsafe reader get in MessageDef

yunhanw-google, 2020-11-20:
So.. this is completely unsafe.  It will do weird and broken things if the value in the TLV is out of range for type T.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/3491#discussion_r527817924_

turon, 2022-01-31:
Referencing this code:

[src/app/MessageDef.cpp]
```
    VerifyOrExit(aTLVType == reader.GetType(), err = CHIP_ERROR_WRONG_TLV_TYPE);

    err = reader.Get(*apLValue);
```

But that file has since been significantly refactored into an entire directory: `src/app/MessageDef`.

Difficult to say whether the issue still exists, but need to follow-up and verify for v1.

Removing `v1_triage_split_4` tag.

bzbarsky-apple, 2022-02-01:
The important part is that we fixed `Get` on the reader to not do broken things in #9944.  So this is good now.

closed 2022-02-01 by bzbarsky-apple