project-chip/connectedhomeip #3992: hotfix label should bypass restyle.io as well.

andy31415, 2020-11-25:
#### Problem
Restyle.io sometimes takes a long time (unsure why) and I am unable to submit even hotfix changes.

 #### Proposed Solution
If hotfix bypasses review, it should be able to bypass restyleio as well.

woody-apple, 2021-02-05:
Done

woody-apple, 2021-02-05:
Oh wait, not done :) back.

woody-apple, 2021-02-05:
```
    ############################################################
    #  Bypass reviews
    ############################################################
    - "'hotfix' not in labels"

    ############################################################
    #  Required status checks
    ############################################################
    - condition: "'*restyle*' in statuses.successful"
      unmet_status: "failure"
      explanation: "Style must be inline before reviewing can be complete"
```

Nevermind, it is for real.

closed 2021-02-05 by woody-apple