project-chip/connectedhomeip #4003: Android project is not compiling

virupaksha, 2020-11-26:
I checked out android source code but it seems code is not compiling. 
[image]

Damian-Nordic, 2020-12-02:
Have you walked through https://github.com/project-chip/connectedhomeip/blob/master/src/android/CHIPTool/README.md before building the app in Android Studio? The error suggests that you don't have CHIP libraries built.

khushbushah2302, 2020-12-21:
Facing below error while running `TARGET_CPU=arm64 ./scripts/examples/android_app.sh` command.

  Setting up CIPD package manager...done
  Setting up Python environment.....[|]
ERROR at //build/config/linux/pkg_config.gni:97:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /Users/myuser/connectedhomeip/.environment/pigweed-venv/connectedhomeip-62ae70af65fcad023af5684f20c857b2/
Command: python3 /Users/myuser/connectedhomeip/build/config/linux/pkg-config.py openssl
Returned 1.
stderr:

Package openssl was not found in the pkg-config search path.
Perhaps you should add the directory containing `openssl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openssl' found
Could not run pkg-config.

See //src/crypto/BUILD.gn:39:3: whence it was called.
  pkg_config("openssl_config") {
  ^-----------------------------
See //BUILD.gn:50:7: which caused the file to be included.
      "${chip_root}/src/crypto",
      ^------------------------

msandstedt, 2020-12-29:
Build also seems to have broken in one of the in-house prototypes at SmartThings, which brings in connectedhomeip as an external cmake project.  This was exporting openssl with OPENSSL_DIR.  This broke for us sometime between Nov 1 and Dec 14.  We are working to change this to instead export PKG_CONFIG_PATH.

I wonder if PKG_CONFIG_PATH is not correctly set for this android build.

woody-apple, 2021-02-02:
Is this still an issue? @mspang ?

virupaksha, 2021-03-23:
Facing error when I run "TARGET_CPU=arm64 ./scripts/examples/android_app.sh".

 Below are the logs which I see at the end. Can someone help me to solve this issue?

//*********** Logs **************//
+ ninja -C out/android_arm64 src/setup_payload/java src/controller/java default
ninja: Entering directory `out/android_arm64'
ninja: no work to do.
+ rsync -a out/android_arm64/lib/CHIPController.jar out/android_arm64/lib/SetupPayloadParser.jar src/android/CHIPTool/app/libs
+ rsync -a out/android_arm64/lib/jni/arm64-v8a src/android/CHIPTool/app/src/main/jniLibs
+ git submodule update --init --recursive third_party/ot-commissioner/repo
+ TARGET_CPU=arm64
+ ./third_party/ot-commissioner/build-android-libs.sh
./third_party/ot-commissioner/build-android-libs.sh: line 3: realpath: command not found
CMake Error: The source directory "/Users/virupakshfutane/Desktop/CHIP/repo" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.

//***********************//

hainahouwei, 2021-04-17:
> I have checked android source code. Could you please help me to compile this source this code. PLease find the attached screenshot for the error what I face during compile.
> 
> [image]

Have you fix this problem ? If so ,can you plese tell me how to fix it?

virupaksha, 2021-04-17:
@hainahouwei  Yes I am able to compile & run the code. :).  Follow this link https://github.com/project-chip/connectedhomeip/blob/master/docs/BUILDING.md as a step 1 & this link https://github.com/project-chip/connectedhomeip/tree/master/src/android/CHIPTool as step 2.

hainahouwei, 2021-04-19:
> @hainahouwei Yes I am able to compile & run the code. :). Follow this link https://github.com/project-chip/connectedhomeip/blob/master/docs/BUILDING.md as a step 1 & this link https://github.com/project-chip/connectedhomeip/tree/master/src/android/CHIPTool as step 2.

@virupaksha Thank you for you reply.
yes, these two step ,and my system is Ubuntu 20.04 LTS.
However ,I facing a  new error as follow ,when I Click Make Project in Android Studio.

` Unresolved reference: KeyValueStoreManager`

When  I run the following command in the command line:
it print error as "/usr/bin/env: ‘sh\r’: No such file or directory"

So, I want to know what system do you use?
Do you build the project on   Android Studio  or command line ?

virupaksha, 2021-04-20:
@hainahouwei  I am using Mac OS for both Android & iOS project. And yes on Android studio I build the project. 
Could you please clean project >> then rebuild the project. Or "Sync Project with Gradle Files".

closed 2021-05-05 by virupaksha