project-chip/connectedhomeip #4061: chip tool not seeing BT adv on Raspi running with Ubuntu desktop linux.

cjandhyala, 2020-12-02:
Chip tool not pairing over BLE on raspi running with Ubuntu desktop linux.

./out/host/chip-tool pairing ble SSID_XXX PASSWD_YYY 12345678 1383
CHIP: [DL] _Init
CHIP: [IN] TransportMgr initialized
CHIP: [IN] local node id is 112233

CHIP: [DL] CHIP task running
CHIP: [BLE] NewConnection
CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device: @
CHIP: [TOO] Pairing Failure: Ble Error 6001 (0x00001771): Incorrect state
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller

/connectedhomeip/examples/lighting-app/linux$ sudo out/host/chip-tool-server 
[sudo] password for pi: 
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*K2H5+B9SS0]
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321

CHIP:DL: wpa_supplicant: connected to wpa_supplicant proxy
CHIP:DL: wpa_supplicant: WiFi interface: /fi/w1/wpa_supplicant1/Interfaces/0
CHIP:DL: Avahi client registered
CHIP:IN: New pairing for device 112233, key 0!!
CHIP:DIS: setup mdns service
CHIP:DL: PublishService 4873986923C64567._chipc._udp11097.-1
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bus acquired for name CHIP-1383
CHIP:DL: CREATE service object at /chipoble/e2e2/service
CHIP:DL: wpa_supplicant: connected to wpa_supplicant interface proxy
CHIP:DL: Create characteristic object at /chipoble/e2e2/service/c1
CHIP:DL: Create characteristic object at /chipoble/e2e2/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/e2e2/service
CHIP:DL: CHIP BTP C2 /chipoble/e2e2/service
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: Avahi group established
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent , 16387
CHIP:DL: Create adv object at /chipoble/e2e2/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x67, 0x05, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent , 16388
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent , 16389
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
CHIP:DL: HandlePlatformSpecificBLEEvent , 8

erjiaqing, 2020-12-03:
There is an issue of ubuntu desktop, a workaround for this issue is to plug a usb bluetooth dongle to the raspberry pi, and then use ` --ble-device <device number>` to specify the device to use.

For example, run `hciconfig` on raspbeery pi to get a list of bluetooth adapters:
```
➜  hciconfig
hci0:   Type: Primary  Bus: UART
    <...>
hci1:   Type: Primary  Bus: USB
    <...>
```

Then you can append ` --ble-device 1` to the command line to use `hci1` device.

andy31415, 2020-12-03:
> There is an issue of ubuntu desktop, a workaround for this issue is to plug a usb bluetooth dongle to the raspberry pi, and then use ` --ble-device <device number>` to specify the device to use. [...]

Is this limitation something we expect to be able to overcome eventually or is this an always required workaround?

cjandhyala, 2020-12-03:
I tried the said option, having BT dongle, but still chip tool doesn't see it. 

$ hciconfig
hci0:	Type: Primary  Bus: USB
	BD Address: BC:14:EF:A2:F6:E8  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:1882 acl:11 sco:0 events:100 errors:0
	TX bytes:2910 acl:11 sco:0 commands:95 errors:0

 ./out/host/chip-tool pairing ble XXX YYY 12345678 1383
CHIP: [DL] _Init
CHIP: [IN] TransportMgr initialized
CHIP: [IN] local node id is 112233

CHIP: [DL] CHIP task running
CHIP: [BLE] NewConnection
CHIP: [BLE] CBManagerState: ON
CHIP: [BLE] Connecting to device: @
CHIP: [TOO] No response from device
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller

sudo out/host/chip-tool-server --ble-device 0
SetupPINCode: [12345678]
SetupQRCode:  [CH:I34DV*K2H5+B9SS0]
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 12344321

CHIP:DL: wpa_supplicant: connected to wpa_supplicant proxy
CHIP:DL: wpa_supplicant: WiFi interface: /fi/w1/wpa_supplicant1/Interfaces/0
CHIP:DL: Avahi client registered
CHIP:DL: wpa_supplicant: connected to wpa_supplicant interface proxy
CHIP:IN: New pairing for device 112233, key 0!!
CHIP:DIS: setup mdns service
CHIP:DL: PublishService 4873986923C64567._chipc._udp11097.-1
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bus acquired for name CHIP-1383
CHIP:DL: CREATE service object at /chipoble/fc57/service
CHIP:DL: Create characteristic object at /chipoble/fc57/service/c1
CHIP:DL: Create characteristic object at /chipoble/fc57/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/fc57/service
CHIP:DL: CHIP BTP C2 /chipoble/fc57/service
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: Avahi group established
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent , 16387
CHIP:DL: Create adv object at /chipoble/fc57/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x67, 0x05, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent , 16388
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: FAIL: RegisterAdvertisement : Timeout was reached
CHIP:DL: HandlePlatformSpecificBLEEvent , 16389
CHIP:DL: Disabling CHIPoBLE service due to error: Error 4003 (0x00000FA3)

vivien-apple, 2020-12-18:
With tip `chip-tool` can connect to the device, but it seems to be an issue if the `Bluez` code at: https://github.com/project-chip/connectedhomeip/blob/316a6f5677c6d53ba91c962b308e0c3886e74dad/src/platform/Linux/CHIPBluezHelper.cpp#L363

`read` seems to fail and returns `0`. 

Here a log with a slightly modified version of read in order to have a more informative error (and we end up with a not very helpful `EOF`)

```
CHIP:DL: Avahi connecting
CHIP:BLE: Waiting for PBKDF param request
CHIP:DIS: CHIP minimal mDNS configured as 'Operational device'.
CHIP:DIS: CHIP minimal mDNS started advertising.
CHIP:IN: New pairing for device 112233, key 0!!
CHIP:SVR: Server Listening...
CHIP:DL: TRACE: Bus acquired for name CHIP-1383
CHIP:DL: CREATE service object at /chipoble/075e/service
CHIP:DL: Create characteristic object at /chipoble/075e/service/c1
CHIP:DL: Create characteristic object at /chipoble/075e/service/c2
CHIP:DL: CHIP BTP C1 /chipoble/075e/service
CHIP:DL: CHIP BTP C2 /chipoble/075e/service
CHIP:DL: TRACE: Bluez mainloop starting BluezMainLoop
CHIP:DL: PlatformBlueZInit init success
CHIP:DL: Long dispatch time: 1000 ms
CHIP:DL: BluezPeripheralRegisterAppDone done
CHIP:DL: HandlePlatformSpecificBLEEvent , 16387
CHIP:DL: Create adv object at /chipoble/075e/advertising
CHIP:DL: SET service data to {'0xFEAF': <[byte 0x00, 0x67, 0x05, 0x5a, 0x23, 0xff, 0xfe]>}
CHIP:DL: HandlePlatformSpecificBLEEvent , 16388
CHIP:DL: CHIPoBLE advertising config complete
CHIP:DL: RegisterAdvertisement complete
CHIP:DL: HandlePlatformSpecificBLEEvent , 16389
CHIP:DL: Bluez coonnected
CHIP:DL: Device A4:83:E7:E3:E4:55 (Path: /org/bluez/hci1/dev_A4_83_E7_E3_E4_55) Connected
CHIP:BLE: CHIPoBluez_NewConnection: 0xffff9c00d430
CHIP:DL: BluezCharacteristicAcquireWrite is called, conn: 0xffff9c00d430
CHIP:DL: c1 BluezCharacteristicWriteFD mtu, 515
CHIP:DL: EOF
CHIP:DL: FAIL: short read in BluezCharacteristicWriteFD (0)
CHIP:DL: CHIPoBLE subscribe received
CHIP:BLE: no endpoint for sub recvd
CHIP:DL: HandlePlatformSpecificBLEEvent , 4
CHIP:DL: HandlePlatformSpecificBLEEvent , 32777
CHIP:DL: Receive kCHIPoBLEConnectionEstablished
CHIP:DL: Bluez disconnected
CHIP:DL: Bluez notify CHIPoBluez connection disconnected
CHIP:DL: HandlePlatformSpecificBLEEvent , 8
```

For what it worth I've tried to connect with and without a dongle. In both cases `chip-tool` connects to the device over bluetooth but the connection fails because of this error.

saurabhst, 2020-12-18:
> There is an issue of ubuntu desktop, a workaround for this issue is to plug a usb bluetooth dongle to the raspberry pi, and then use ` --ble-device <device number>` to specify the device to use. [...]

@erjiaqing if the above set up configuration is proved to be absolute must for things to get working , would it make sense to put this piece of information in the README text for the chip-tool if not already ?

wissn84, 2021-01-28:
Hi, 
I have meet the same issue related to "short read error" although our RP use BT Dongle and set ble device option.
Source code version is 9bcb03eab19c37471be12c09fa09d568c6864ff7.

cjandhyala, 2021-02-04:
Chip tool is not POR anymore. We will use python chip dev ctrl in place of chip-tool

closed 2021-02-04 by cjandhyala