project-chip/connectedhomeip #4062: chip tool could not get response from ESP32 devices

cjandhyala, 2020-12-02:
chip tool is pairing with device over BLE, getting IP address. But not receiving responses from M5.

 ./out/host/chip-tool basic discover 1
CHIP: [DL] _Init
CHIP: [IN] TransportMgr initialized
CHIP: [IN] local node id is 112233

CHIP: [DL] CHIP task running
CHIP: [TOO] Endpoint id: '0x01', Cluster id: '0x0000', Command id: '0x0c'
CHIP: [ZCL] Successfully encoded 11 bytes
CHIP: [DL] select failed: OS Error 2009 (0x000007D9): Bad file descriptor

CHIP: [IN] New pairing for device 12344321, key 0!!
CHIP: [IN] Sending msg from 112233 to 12344321
CHIP: [IN] Secure transport transmitting msg 0 after encryption
CHIP: [CTL] SendMessage returned 0
CHIP: [TOO] No response from device
CHIP: [CTL] Shutting down the commissioner
CHIP: [CTL] Shutting down the controller
CHIP: [TOO] Run command failure: Error 4172 (0x0000104C)

host log :

GATT procedure initiated: notify; att_handle=14
I (64260) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (64260) app-devicecallbacks: Server ready at: 192.168.1.103:11097
I (64270) all-clusters-app: Display awake but will switch off automatically in 30 seconds
I (64280) app-devicecallbacks: Current free heap: 112760

GATT procedure initiated: notify; att_handle=14
I (64290) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
GAP procedure initiated: stop advertising.
I (64310) chip[DL]: CHIPoBLE advertising stopped
I (64310) app-devicecallbacks: Current free heap: 112776

I (64380) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
GATT procedure initiated: notify; att_handle=14
I (64390) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
GATT procedure initiated: notify; att_handle=14
I (64400) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (64490) chip[DL]: Write request/command received for CHIPoBLE TX CCCD characteristic (con 0 ) indicate = 0 notify = 0
I (64490) chip[DL]: CHIPoBLE unsubscribe received
I (64490) chip[SVR]: OnRendezvousError: Ble Error 6000004 (0x005B8D84): BLE central unsubscribed
I (64500) chip[DL]: BLE GAP connection terminated (con 32728
I (65080) chip[BLE]: Releasing end point's BLE connection back to application.
E (65080) chip[DL]: Long dispatch time: 594 ms
I (65090) chip[DL]: IP_EVENT_GOT_IP6
I (65090) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:266f:28ff:fe8f:d758
I (65100) app-devicecallbacks: Current free heap: 113668

I (83170) chip[SVR]: Packet received from UDP:192.168.1.115:11097: 17 bytes
I (83170) chip[ZCL]: APS frame processing success!
I (83170) chip[ZCL]: RX len 6, ep 1, clus 0x 0 
I (83180) chip[ZCL]:  FC 0 seq 1 cmd c payload[
I (83180) chip[ZCL]: 00 00 FF 
I (83190) chip[ZCL]: ]
I (83190) chip[ZCL]: 0x3f40dbcd0x3f400d74: clus  0
I (83200) chip[IN]: Sending msg from 12344321 to 112233
I (83200) chip[ZCL]: T   0:TX (0x3f40c130) Ucast 0x00x3f400d74
I (83210) chip[ZCL]: TX buffer: [
I (83210) chip[ZCL]: 08 01 0D 01 00 00 20 07 00 30 FD FF 21 
I (83220) chip[ZCL]: ]
I (83220) chip[ZCL]: Data model processing success!
I (94270) all-clusters-app: Display going to sleep...
I (230320) chip[SVR]: Packet received from UDP:192.168.1.115:11097: 41 bytes
I (230320) chip[SVR]: Client sent: Message from Standalone CHIP echo client!
I (230330) chip[IN]: Sending msg from 12344321 to 112233
I (230340) chip[SVR]: Echo sent

vivien-apple, 2020-12-18:
I have a hard time to reproduce it locally. I've tried to reproduce it on `esp32`, `rpi` and on local builds without success. 

I "think" I've seen this behavior once on the `M5Stack` but only for a short duration of time before packets starts being sent correctly.

The above log shows some interesting infos, particularly this block:
```
I (83200) chip[IN]: Sending msg from 12344321 to 112233
I (83200) chip[ZCL]: T 0:TX (0x3f40c130) Ucast 0x00x3f400d74
I (83210) chip[ZCL]: TX buffer: [
I (83210) chip[ZCL]: 08 01 0D 01 00 00 20 07 00 30 FD FF 21
I (83220) chip[ZCL]: ]
I (83220) chip[ZCL]: Data model processing success!
```

Basically this block tell us that the device has received the message, parsed it successfully and then has called the CHIP stack to send the response and succeed without errors.

I've tried to dig into this code path to see if there is any case where we can call `Transport::SendMessage` and succeed without any errors but without sending the message. So far I've looked until `Inet::UDPEndpoint` without success.

At that point it is not clear to me on which side of the pipe the error happens. I would appreciate if someone who can reproduce it can launch a `Wireshark` or an equivalent to see if the packets are being sent by the device or not, and if this is a local thing that prevents those packets to comes in.

@cjandhyala Can you look at that ?

woody-apple, 2021-02-02:
ping @cjandhyala

cjandhyala, 2021-02-02:
closing the issue as this issue is not observed anymore with the latest chip stack

closed 2021-02-02 by cjandhyala