project-chip/connectedhomeip #4069: Add mDNS for finding device by discriminator in chip device controller

erjiaqing, 2020-12-03:
#### Problem
We does not support select device by discriminator in chip device controller.

 #### Proposed Solution
Implement it after https://github.com/project-chip/connectedhomeip/pull/4068 is merged.

andy31415, 2021-02-22:
Is this for commisionable discovery?

I would expect controllers needing to search by NodeID and not by discriminators as discriminators may not be unique.
Could you extend the description on why the functionality is needed and what the use would be?

erjiaqing, 2021-02-23:
This is used for commissionable discovery. We should be able to get a list of devices with their discriminator / vendorid / product id so they can be used for discovery before service commissioning.

> Is this for commisionable discovery?
> 
> I would expect controllers needing to search by NodeID and not by discriminators as discriminators may not be unique.
> Could you extend the description on why the functionality is needed and what the use would be?

Damian-Nordic, 2021-04-13:
Is it a duplicate of https://github.com/project-chip/connectedhomeip/issues/5715?

closed 2021-07-21 by Damian-Nordic