project-chip/connectedhomeip #4093: Rename member function IsNeedsAck to NeedsAck in payloadHeader

yufengwangca, 2020-12-04:
#### Problem
The convention should be that the code should read like text, ideally. The needs ack case as currently defined ends up reading as "this payload header is needs ack", which is not very good English. "this payload header needs ack" is closer to what one would actually say.

 #### Proposed Solution
Rename member function IsNeedsAck to NeedsAck in payloadHeader

woody-apple, 2021-02-02:
@yufengwangca how does this interact with https://github.com/project-chip/connectedhomeip/issues/4083?

yufengwangca, 2021-02-02:
They are not relevant,  this function indicates if an incoming message needs ack, #4083 indicates if the incoming message contains ack info.

closed 2021-02-09 by saurabhst