project-chip/connectedhomeip #4113: chip tool pairing over BLE fails on linux

cjandhyala, 2020-12-08:
sudo ./out/host/chip-tool pairing ble XXX YYY 012345678 3840
CHIP:IN: TransportMgr initialized
CHIP:IN: local node id is 112233

CHIP:DL: wpa_supplicant: connected to wpa_supplicant proxy
CHIP:DL: wpa_supplicant: WiFi interface: /fi/w1/wpa_supplicant1/Interfaces/1
CHIP:DL: Avahi client registered
CHIP:DL: CHIP task running
CHIP:DL: FAIL: BLE name is NULL in ConfigureBluezAdv
CHIP:DL: Disabling CHIPoBLE service due to error: Error 4003 (0x00000FA3)
CHIP:DL: wpa_supplicant: connected to wpa_supplicant interface proxy
CHIP:TOO: No response from device
CHIP:CTL: Shutting down the commissioner
CHIP:CTL: Shutting down the controller

andy31415, 2021-01-12:
During todays SWTT meeting we proposed to deprecate chip-tool and only support the python wrapper for chip automation going forward.

The assumption for this issue is that BLE central fixes for linux affect the chip library and are not chip-tool specific.

@Damian-Nordic : is this assumption correct? assuming you make chip-tool work today, is it reasonable to assume that this will make the python wrapper work on linux once we move to only python?

Damian-Nordic, 2021-01-13:
@andy31415 I believe the python tool has its own implementation of the BLE central, written in Python (see `ChipBluezMgr.py` file) and CHIP simply forwards all necessary function calls to the python side. It's similar to how Android CHIPTool works. What I have implemented is the BLE central implementation in C++, so it's necessary for native Linux controllers, such as chip-tool. Do you think it's still useful that I integrate my changes even if we (maybe temporarily) won't have any native controllers once we remove chip-tool?

PR #4347 (2021-01-14): [linux] BLE central implementation
 #### Problem
Linux platform layer provides functions for the BLE peripheral, however the BLE central role is only partially implemented. As a result, Linux native controllers, such as `examples/chip-tool` cannot initiate a BLE connection and perform Rendezvous over BLE.
Although `examples/chip-tool` may get deprecated in the future due to #4334, the changes posted here may still be useful if one needs to implement another controller app, or add the controller capabilities to `examples/shell`.

 #### Summary of Changes
* For controller applications, configure [... cut]
closed 2021-01-14 by rwalker-apple