project-chip/connectedhomeip #4143: Full RasPi Support for CHIP

kidip, 2020-12-09:
#### Needs
- CHIP Stack Compilable on the RasPi
- `chip-tool` running
- `chip-tool` commissioning support
- `chip-tool` cluster control support
- Multiple Sample app Types (or easy ability to create them) for RasPi

#### Reasons
Our CHIP Certification Tool is going to be running on the RasPi.  We want to have simulated devices that are running on the RasPi.  We want to use `chip-tool` to do communication with them.

kidip, 2020-12-09:
Some Related Issues:
#4061 #4062

tima-q, 2021-01-13:
Related:
#4334 - Deprecation of chip-tool

woody-apple, 2021-02-02:
Note: Need to somehow break this down into subtasks/projects.

franck-apple, 2021-02-05:
> Note: Need to somehow break this down into subtasks/projects.

Turned the initial list of Needs into subtasks.