project-chip/connectedhomeip #4190: Various opaque contexts in CryptoPAL should ensure they have sufficient alignment

bzbarsky-apple, 2020-12-11:
#### Problem
We should ensure that the alignment requirements of `HashSHA256OpaqueContext` are no lower than the various concrete contexts it represents.  Similar for other opaque contexts in CryptoPAL.

 #### Proposed Solution
Add relevant static asserts if we can.

@pan-apple

andy31415, 2022-08-24:
v1 review: @bzbarsky-apple could you provide an opinion/statement if this should block v1 or not?

andy31415, 2022-08-25:
I believe that `add static asserts` should not be v1blocking. Please adjust @bzbarsky-apple if you disagree.

bzbarsky-apple, 2022-08-26:
It's not v1 blocking if the asserts would pass @andy31415 

If they fail, we are going to have issues with our crypto operations....

bzbarsky-apple, 2022-08-26:
Looks like we started using `SafePointerCast` to cast from the opaque contexts, which enforces the necessary alignment bits via static assert.

closed 2022-08-26 by bzbarsky-apple